mne-tools/mne-python #3330: Gradiometers RMS plots multiple conditions

kambysese, 2016-06-22:
I am trying to compared evoked files using `plot_evoked_topo(evokeds, merge_grads=True)` where `evokeds=[evoA.pick_types(meg='grad'), evoB..pick_types(meg='grad')]`; the shape of data in each evoked obj is 204x421, however `.viz.topo` returns

> ValueError: operands could not be broadcast together with shapes (2,) (421,) @ ln 340

@Eric89GXL according to the documentation `plot_evoked_topo` implies combining grads is possible for an evoked object or list of evokeds. At least that is my interpretation.

agramfort, 2016-06-23:
can you paste a full gist to replicate the pb?

jaeilepp, 2016-06-23:
Reproduced with:
```
import mne
path = mne.datasets.sample.data_path()
raw = mne.io.Raw(path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif')

events = mne.find_events(raw)
epochs  = mne.Epochs(raw, events, 1)
epochs2 = mne.Epochs(raw, events, 2)
evokeds = [epochs.average(), epochs.average()]
mne.viz.plot_evoked_topo(evokeds, merge_grads=True)
```
I'll look into it.

closed 2016-06-24 by jaeilepp