mne-tools/mne-python #3336: mne.time_frequency.psd_welch add median option

sapttrs, 2016-06-24:
I want to robustly estimate a power spectrum from raw data. 
I am concerned that segments of my raw data contain artifacts. 
mne.time_frequency.psd_welch divides the raw data into N (possibly overlapping) windows, computes N psds, and returns the average psd. 
Could we add the option to return the median psd? 
By this I mean for each frequency bin we have N estimates of the power (from the N psds), and we return the median of these N values. 
I believe the R function welchPSD does this.

agramfort, 2016-06-26:
+1 for a method = "mean" | "median" as the R implem.

do you want to take a stab at implementing this?

kingjr, 2016-06-26:
Would it make sense to also make `method` accept a float between .5 and 1, in order to indicate the percentile of data from which the average is computed? `1` would be equivalent to the mean, `.5` would be equivalent to the median, and the rest would be trimmed means, which in my experience are more stable than median in our low-sampled datasets.

kingjr, 2016-06-26:
(also see https://github.com/mne-tools/mne-python/issues/3044)

agramfort, 2016-06-26:
yes I think that trimmed mean would be nice.

See API of scipy:

http://docs.scipy.org/doc/scipy-0.17.0/reference/generated/scipy.stats.trim_mean.html

let's try to copy it as much as possible.

sapttrs, 2016-06-27:
I will take a stab at implementing. 
I agree a trimmed mean method is best. 
After looking at the code in more detail, I believe mne calls scipy.signal.welch to compute the psd, so it is scipy that I should attempt to contribute to first (to get scipy.signal.welch to return a trimmed mean).

Eric89GXL, 2016-11-18:
@sapttrs do you have time to look into this? If not I can try

dengemann, 2016-11-18:
If we go this rout consider the discussion on the other recent issue about
Annotations. We then should also think of conputing the welch epochs
explicitly to be able to compute stats across windows, e.g, for estimating
variability/confidence in spectral features.

sapttrs, 2016-11-21:
@Eric89GXL I haven't had time to look into it, but a student at our lab has worked on it. I will get an update from him.

fkamel, 2016-11-22:
@Eric89GXL @sapttrs scipy's welch function (scipy/signal/spectral.py) can be easily modified to implement trimmed mean. However, mne's psd_welch (mne/time_frequency/psd.py) uses parallel calls to welch to compute the psds, so the way to do it seems to be by modifying `psds' at the end of _psd_welch.

fkamel, 2016-11-30:
By adding a parameter tmean with default argument = 0 to both psd_welch and _psd_welch, the following code added to the end of _psd_welch seems to do the job. It can be modified to support parallel processing as well.

   

    if tmean > 0:
        from scipy import stats
        median_window = np.zeros(psds.shape[1])   
        for j in range(psds.shape[1]): 
            # process all spectral densities at element j of each window.
            median_window[j] = stats.trim_mean(psds[:,j], tmean)
        return median_window, psds, freqs                       
    else:
        return psds, freqs

agramfort, 2016-12-01:
@fkamel is the median computed over channels or over windows? From what I understand you want it over windows

larsoner, 2019-08-20:
Marking this as easy now that we have https://github.com/scipy/scipy/pull/8970

PR #6748 (2019-09-15): ENH: Add support for median averaging and no averaging to Welch PSD
#### Reference issue
Closes #3336.

#### What does this implement/fix?

- Adds an `average` kwarg to `psd_welch()` and `psd_array_welch()`:
  - `average='mean'`: returns PSD based on the arithmetic mean of all segments, calculated via `scipy.signal.welch()`
  - `average='median'`: returns PSD based on the median of all segments, corrected for bias relative to the mean, calculated via `scipy.signal.welch()`
  - `average=None`: returns all segments unaggregated, calculated via `scipy.signal.spectrogram()`

- <strike>There is cur [... cut]
closed 2019-09-15 by agramfort