mne-tools/mne-python #3338: mne.io.set_bipolar_reference - info channel name inconsistency

klanderson, 2016-06-26:
Hi, I'm trying to get bipolar rereferenced eog channels and am encountering an error:

`RuntimeError: info channel name inconsistency detected, please notify mne-python developers`

Here is the full trace:

```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-2-b625a9834e8b> in <module>()
     18 
     19 # Bipolar reference EOG channels
---> 20 mne.io.set_bipolar_reference(raw, ['LHEOG','TVEOG'], ['RHEOG','BVEOG'], ch_name=['HEOG','VEOG'], copy=False)
     21 
     22 # Add Cz back in to data. It was used as reference

/Users/xxxx/anaconda/envs/xxxx/lib/python3.5/site-packages/mne/io/reference.py in set_bipolar_reference(inst, anode, cathode, ch_name, ch_info, copy)
    371     # Perform bipolar referencing
    372     for an, ca, name, info in zip(anode, cathode, ch_name, new_ch_info):
--> 373         _apply_reference(inst, [ca], [an])
    374         an_idx = inst.ch_names.index(an)
    375         inst.info['chs'][an_idx] = info

/Users/xxxx/anaconda/envs/xxxx/lib/python3.5/site-packages/mne/io/reference.py in _apply_reference(inst, ref_from, ref_to)
     69                            'preload=True (or string) in the constructor.')
     70 
---> 71     eeg_idx = pick_types(inst.info, eeg=True, meg=False, ref_meg=False)
     72 
     73     if ref_to is None:

/Users/xxxx/anaconda/envs/xxxx/lib/python3.5/site-packages/mne/io/pick.py in pick_types(info, meg, eeg, stim, eog, ecg, emg, ref_meg, misc, resp, chpi, exci, ias, syst, seeg, dipole, gof, bio, ecog, include, exclude, selection)
    250         raise TypeError('info must be an instance of Info, not %s'
    251                         % type(info))
--> 252     info._check_consistency()
    253     nchan = info['nchan']
    254     pick = np.zeros(nchan, dtype=np.bool)

/Users/xxxx/anaconda/envs/xxxx/lib/python3.5/site-packages/mne/io/meas_info.py in _check_consistency(self)
    246                 ch_1 != ch_2 for ch_1, ch_2 in zip(self['ch_names'], chs)) or \
    247                 self['nchan'] != len(chs):
--> 248             raise RuntimeError('info channel name inconsistency detected, '
    249                                'please notify mne-python developers')
    250 

RuntimeError: info channel name inconsistency detected, please notify mne-python developers
```

And here is my code:
```

# Read all three files and append together into one dataset
data_path = './data/20160608A/'
vhdr_fname = op.join(data_path, '20160608A_02.vhdr')
raw = mne.io.read_raw_brainvision(vhdr_fname, 
                            montage=None, misc=(), preload=True,
                            response_trig_shift=0, event_id=None, verbose=0)
vhdr_fname = op.join(data_path, '20160608A_04vhdr')
raw2 = mne.io.read_raw_brainvision(vhdr_fname, 
                            montage=None, misc=(), preload=True,
                            response_trig_shift=0, event_id=None, verbose=0)
vhdr_fname = op.join(data_path, '20160608A_06.vhdr')
raw3 = mne.io.read_raw_brainvision(vhdr_fname, 
                            montage=None, misc=(), preload=True,
                            response_trig_shift=0, event_id=None, verbose=0)
raw.append([raw2, raw3], preload=True)
raw2 = None
raw3 = None

# Bipolar reference EOG channels
mne.io.set_bipolar_reference(raw, ['LHEOG','TVEOG'], ['RHEOG','BVEOG'], ch_name=['HEOG','VEOG'], copy=False)
```

klanderson, 2016-06-26:
The issue has something to do with trying to do multiple pairs at once. I changed

```
mne.io.set_bipolar_reference(raw, ['LHEOG','TVEOG'], ['RHEOG','BVEOG'], ch_name=['HEOG','VEOG'], copy=False)
```

to
```
mne.io.set_bipolar_reference(raw, 'LHEOG', 'RHEOG', ch_name='HEOG', copy=False)
mne.io.set_bipolar_reference(raw, 'TVEOG', 'BVEOG', ch_name='VEOG', copy=False)
```

and the code works fine.

agramfort, 2016-06-26:
@wmvanvliet is that expected?

wmvanvliet, 2016-06-26:
no, this is definitely a bug. Either in `set_biopolar_reference` or in `info._update_redundant`. I will investigate.

PR #3343 (2016-06-27): Fix bug in set_bipolar_reference
Closes #3338

There was a test missing for setting multiple references with the `mne.io.set_bipolar_reference` function. With the back and forth on the `Info` object, this case therefore slipped through the regression testing. This PR adds the test and fixes the code.
commit 93e36965 (2016-06-27): Fix bug in set_bipolar_reference (#3343)

* Fix bug in set_bipolar_reference

Closes #3338

There was a test missing for setting multiple references with the
`mne.io.set_bipolar_reference` function. With the back and forth on the
`Info` object, this case therefore slipped through the regression
testing. This PR adds the test and fixes the code.

* remove debug prints

* PEP8
closed 2016-06-27 by agramfort