mne-tools/mne-python #3347: AtrributeError while reading EEGLab data

pfaion, 2016-06-27:
Hi,
I am trying to read in some data from EEGLab with the function
[mne.io.read_raw_eeglab()](https://martinos.org/mne/stable/generated/mne.io.read_raw_eeglab.html#mne.io.read_raw_eeglab)

but I get the error
**AttributeError: 'int' object has no attribute 'isdigit'**
[in line 541 of io/eeglab/eeglab.py](https://github.com/mne-tools/mne-python/blob/master/mne/io/eeglab/eeglab.py#L541)

Any ideas on this?

jaeilepp, 2016-06-27:
Perhaps it needs to be cast to string on https://github.com/mne-tools/mne-python/blob/master/mne/io/eeglab/eeglab.py#L533 @mmagnuski @jona-sassenhagen ?
Just a stab in the dark...

jona-sassenhagen, 2016-06-27:
Yeah, @pfaion is this with current MNE master?

agramfort, 2016-06-27:
@pfaion if it's still broken with mne master please share a file

jona-sassenhagen, 2016-06-27:
I think the line @jaeilepp links to would actually fail if you have a file created with an old version of EEGLAB with only a single event, so we need to fix either way.

But let's first see if this is the problem @pfaion has run into, or if there's another problem.

pfaion, 2016-06-28:
Works with the current master. Sorry for any inconvenience...

closed 2016-06-28 by pfaion

jona-sassenhagen, 2016-06-28:
It's good, it made me realize another error.