mne-tools/mne-python #3351: Time representation in SourceEstimate.plot()

christianbrodbeck, 2016-06-27:
`stc.plot()` currently supplies time in ms, while PySurfer `Brain.save_movie()` assumes time is in s. Consequently, the default values as in `stc.plot().save_movie(fname)` produce a very slow movie. A simple solution to make it more intuitive would be to change time to seconds (it can still be labeled in the plot as ms), although that would change the behavior of existing scripts that compensate for the difference with something like `stc.plot().save_movie(fname, time_dilation=0.004)`... I don't know if there is a better solution but feel it would be nice if the default values would lead to a reasonable movie...

agramfort, 2016-06-28:
stc.plot().save_movie(fname)

should work out the box. Just make it work :)

christianbrodbeck, 2016-06-28:
I think the more parsimonious option would be to have `SourceEstimate.plot()` use [s]. Can we justify breaking some old code which assumes brain time is in milliseconds?

An alternative would be to add a flag to `surfer.Brain` that remembers whether time is in [s] or [ms] and adapts the time dilation accordingly.

Preferences?

Eric89GXL, 2016-06-28:
We can add a `time_unit=None` kwarg for deprecation, 0.13 it means `ms` 0.14 we change it to `s`

christianbrodbeck, 2016-06-28:
@Eric89GXL I'll work on that.

A second issue: Since the `config_opts` parameter has been deprecated in PySurfer I was trying to update `plot_source_estimate` accordingly and noticed that that function already seems to be dealing with multiple PySurfer versions... It seems to me it would make things easier and more transparent to just require the latest version of PySurfer, or is there a reason why some people could not upgrade?

Eric89GXL, 2016-06-28:
I'm fine with bumping the minimum required version, it's an easy upgrade and has been out for a while IIRC

closed 2016-07-06 by christianbrodbeck