mne-tools/mne-python #3356: Cross spectral-density estimation for ndararys

nfoti, 2016-06-28:
It looks like the only way to estimate the cross spectral-density (CSD) is via the `compute_epochs_csd` function that requires your data to be stored in an `Epochs` object. It would be very convenient to also be able to compute the CSD of a bunch of trials stored in an `n_trials x n_{channels or labels} x n_time` `ndarray`.

It doesn't look like the current code would be hard to modify since most of the code is just extracting data from the `Epochs` objects and the meat of the code looks generic for `ndarrays`.

Eric89GXL, 2016-06-28:
I agree this would be useful. Can you just tack onto #3355 the refactoring? Try to make the new function names and arguments consistent with the `psd_*` functions, so I guess it might be `csd_multitaper`?

nfoti, 2016-06-28:
Moving discussion to #3355.

sviter, 2017-05-31:
Should this issue be closed?

agramfort, 2017-05-31:
I think so

@nfoti ok?

nfoti, 2017-06-01:
Yes, fine with me.

closed 2017-06-01 by Eric89GXL