mne-tools/mne-python #3357: ENH: Auto-build examples or tutorials that are modified

Eric89GXL, 2016-06-28:
Fun project for someone if they get bored. CircleCI currently builds full docs `html_dev` on `master` and `html_dev-noplot` on PRs. But really on PRs it should build `PATTERN=... make html_dev-pattern` when the PR touches an example or tutorial. So some enterprising person could inspect the `diff` and figure out if any of the files live in `examples/` or `tutorials/` and call the pattern version.

@jaeilepp fancy a challenge? :)

jasmainak, 2016-06-28:
I'm not super sure I agree with this because in principle you could break an example from a diff not in the example ... to me, CircleCI is also a testing mechanism

Eric89GXL, 2016-06-28:
Unfortunately it is not feasible to run all examples for every PR from a
build time standpoint, so this is probably the best we can do for now

agramfort, 2016-06-29:
let's keep this for a sunday hack

jaeilepp, 2016-06-29:
> I'm not super sure I agree with this because in principle you could break an example from a diff not in the example ... to me, CircleCI is also a testing mechanism

But this change wouldn't remove the ``dev-noplot`` build. It would just add building of the modified examples/tutorials. Right?

Eric89GXL, 2016-06-29:
> But this change wouldn't remove the dev-noplot build. It would just add building of the modified examples/tutorials. Right?

It would replace the `make html_dev-noplot` with `make html_dev-pattern`, thus making the testing (and usefulness in reviews) more complete than it is now.

mmagnuski, 2016-06-29:
I have a slightly related question:
when are the dev online docs updated (I cannot see any changes there)?
Or - how can I see the cutting edge docs without building them on my computer? I don't know how to get to relevant circle-artifacts...

Eric89GXL, 2016-06-29:
> when are the dev online docs updated (I cannot see any changes there)?

https://mne-tools.github.io/dev should be updated on every successful build on `master` (i.e., after every PR merge), but it looks like that isn't happening. @jaeilepp any ideas why not?

> how can I see the cutting edge docs without building them on my computer? I don't know how to get to relevant circle-artifacts...

Do you see the status updates from CIs about the builds? One of them is from Circle. Can you try navigating to that one to see if you can see the "artifacts" tab toward the top? Like here:

https://circleci.com/gh/mne-tools/mne-python/1418#artifacts

mmagnuski, 2016-06-29:
I don't see any artifacts tab under the link you posted :(
I see "collect artifacts" section - but there is a lot of text.

Eric89GXL, 2016-06-29:
You don't see something like this:

[image]

What do you see?

mmagnuski, 2016-06-29:
I see this:
[image]

Is creating an account on circle sufficient for artifacts to show up (funny, I usually try to avoid artifacts ;) )?

mmagnuski, 2016-06-29:
Ok, I signed up and I can see the artifacts tab :)

Eric89GXL, 2016-06-29:
Yeah you should just be able to sign in with GitHub

jaeilepp, 2016-06-30:
> @jaeilepp any ideas why not?

I see that the last push was on 11th of May, so [this](https://github.com/mne-tools/mne-python/commit/38aac46012694c941e8255f2e1f7476c7b51c000) probably broke it. I'm not sure why.

jaeilepp, 2016-06-30:
Now it gets updated again.

PR #3365 (2016-07-01): MRG: Build relevant examples
Closes #3357.

Today is Sunday right @agramfort? :)
closed 2016-07-01 by agramfort