mne-tools/mne-python #3358: ENH: Tutorial on coreg

Eric89GXL, 2016-06-28:
@christianbrodbeck it occurred to me it might be nice to have a tutorial on using the coreg GUI. A lot of it would be text, but you could at least set up the inputs, have a set of text instructions, and then show the outputs that one would get. Is that feasible? Any time/interest in doing it?

It would also be cool if there were one for using a surrogate MRI, maybe as a separate tutorial.

christianbrodbeck, 2016-06-28:
Yes completely agree, I was recently thinking of making one in the style of the slides, with annotated screenshots and descriptions, would that fit in?

Eric89GXL, 2016-06-28:
The more scripted you can make it the better, but yeah annotated images
should work too. You should be able to include them somehow.

agramfort, 2016-06-29:
or a nice screencast on an MNE youtube channel? :)

dengemann, 2016-06-29:
I think Christian in fact did such a thing, at least a slide show. Eric is
video experienced afaik.

christianbrodbeck, 2016-07-01:
Yes @dengemann  we had the slides at http://www.slideshare.net/mne-python/mnepython-coregistration and http://www.slideshare.net/mne-python/mnepython-scale-mri. They're linked form the doc string of `gui.coregistration`. I could try to add a similar, image-heavy tutorial to he docs to make the more accessible.

dengemann, 2016-07-01:
+100 for screen cast :)

christianbrodbeck, 2016-07-02:
How old am I to prefer written docs :P

jona-sassenhagen, 2016-07-18:
I'd prefer written docs with pics actually.

Eric89GXL, 2016-07-18:
Me too, easier to reference and link, etc.

kingjr, 2016-07-18:
Also +1 for written docs, so that we can copy/paste and go at the adequate speed.

christianbrodbeck, 2016-07-20:
A related question, I just tried to find out what exactly fsaverage is an average of, and could not find a clear description... https://surfer.nmr.mgh.harvard.edu/fswiki/FsAverage 
refers to a "README file in $FREESURFER_HOME/subjects/fsaverage" but I don't seem to have that readme file. A guess based on the link at the top of the WIKI page, is FSAverage based on the "Buckner40"? (https://surfer.nmr.mgh.harvard.edu/fswiki/Buckner40Testing) Does somebody here know?

Eric89GXL, 2016-07-20:
http://openwetware.org/wiki/Beauchamp:FSAverageReadme

But even from this I'm not sure, maybe there is a trail you can follow in
there

christianbrodbeck, 2016-07-20:
Thanks @Eric89GXL ! `make_average_surface.log` indeed references the [Buckner40 subjects](https://surfer.nmr.mgh.harvard.edu/fswiki/Buckner40Notes)

Eric89GXL, 2016-07-20:
Please add notes to our doc for the next person who is curious

Eric89GXL, 2017-02-14:
Closing for #1495

closed 2017-02-14 by Eric89GXL