mne-tools/mne-python #3362: Tutorial for migrating from Fieldtrip

jona-sassenhagen, 2016-06-29:
We have a tutorial for EEGLAB users. We should have the same for Fieldtrip users, as Fieldtrip is I think the second-largest software platform (and IME a bit more popular amongst MEG people).

I can't do it, I've never really used Fiedltrip.

jona-sassenhagen, 2016-06-29:
https://www.martinos.org/mne/stable/manual/migrating.html EEGLAB tutorial here.

mmagnuski, 2016-06-30:
I've used FieldTrip mainly for time-frequency and cluster-correction, so I could try creating some examples of migration with respect to this. I wouldn't be able to do this soon however.
Currently the table for migrating from EEGlab should work well also for other migrations - one should just ignore the "EEGLAB function" column.

mmagnuski, 2016-06-30:
Related: #3201

Eric89GXL, 2017-02-14:
Closing for #1495

closed 2017-02-14 by Eric89GXL