mne-tools/mne-python #3366: BUG: CircleCI cache problem

Eric89GXL, 2016-06-29:
@jaeilepp I noticed that CircleCI doesn't actually cache `/home/ubuntu/mne_data`, leading to lots of re-downloading. Do you know why?

Maybe we should try caching individual directories like `mne_data/MNE-sample-data` or so. WDYT?

jaeilepp, 2016-06-30:
No idea. I remember having a similar problem when I was starting up the circle builds, but then it magically fixed itself.
> Maybe we should try caching individual directories like mne_data/MNE-sample-data or so. WDYT?

I think it's worth a try.

jaeilepp, 2016-07-01:
It still doesn't explain why the cache doesn't work for our master. At least part of the cache now works for my fork.

Eric89GXL, 2016-07-01:
The most recent `master` build is the one where you told it not to use a cache. We haven't merged anything since then.

jaeilepp, 2016-07-01:
Yes. We should try  merging something :)

jaeilepp, 2016-07-03:
I think this is still an issue on our master branch.

jaeilepp, 2016-07-03:
This seems weird. The master build [here](https://circleci.com/gh/mne-tools/mne-python/1470) seems to use cache from pull [3377](https://circleci.com/gh/mne-tools/mne-python/1476), which is empty because unmerged pull requests don't have the data present. I wonder if we can somehow tell the build which cache to use.

jaeilepp, 2016-07-03:
So it seems that the caching is working, but is lost since the master builds aren't using them.

Eric89GXL, 2016-07-03:
We could try deleting the existing CircleCI integration on the web
interface (I think it wipes all history and associated information) and
then re-enable it

jaeilepp, 2016-07-08:
It seems to have caught sample and bst_auditory in some of the merges. See https://circleci.com/gh/mne-tools/mne-python/1604.
Maybe we can populate the cache now by making some small changes to some examples and merging it after it has cached the stuff.

Eric89GXL, 2016-07-08:
I noticed that, I guess it's fixed now (?!).

closed 2016-07-08 by Eric89GXL

jaeilepp, 2016-07-08:
Well, depends how you look at it. I think eventually it will get populated, but until then, it will keep downloading the other datasets on every merge.

Eric89GXL, 2016-07-08:
Ahh, I see what you mean. So you suggest to make a PR that adds the last dataset checks and touches the examples that use them, and then merge it? Fine by me.

jaeilepp, 2016-07-08:
Yeah, it's not the cleanest of solutions, but just to see if it works.