mne-tools/mne-python #3370: BUG: simulate_evoked errors with finite SNR

nfoti, 2016-06-30:
I am running into a problem where calling `simulate_evoked` with a finite and positive SNR results in an error. The forward solution I am using has 306 channels but only 305 channels of noise are simulated.

This [gist](https://gist.github.com/nfoti/22158db5eb67d447e24d2d07728189a5) demonstrates the problem.

Eric89GXL, 2016-06-30:
Can you make that gist a minimal example? It will help isolate the problem to the fewest calls possible and thus be eaiser to fix

jaeilepp, 2016-07-01:
The problem seem to lie in bad channels being dropped when computing covariance matrix, resulting in evoked and covariance instances having different set of channels. Maybe the correct fix would be just improving the error message and saying that some channels are not present in the covariance instance.

closed 2016-07-04 by agramfort