mne-tools/mne-python #3378: dev docs updated partially?

mmagnuski, 2016-07-01:
In [this docs tutorial](http://martinos.org/mne/dev/auto_tutorials/plot_artifacts_correction_ica.html) the figure have been updated but the text not. The result is a little confusing :)

mmagnuski, 2016-07-01:
If you download the tutorial as `.py` or `.ipynb` - you can see that the text is actually different than the one rendered.

agramfort, 2016-07-02:
is it better now?

mmagnuski, 2016-07-02:
It's better - but now it shows the version from before merging the PR that corrects for bad component inds.

Eric89GXL, 2016-07-02:
It takes over an hour (plus queue time) to build the docs, so it's possible
your PR's changes were (or are) still pending. You can check the CircleCI
site to be sure.

mmagnuski, 2016-07-02:
I checked circle and I could not see any unfinished "circles" [here](https://circleci.com/gh/mne-tools/mne-python) but the dev docs are still behind (although the circle artifacts for my PR were correct).

Eric89GXL, 2016-07-02:
Someone will need to go through the CirceCI logs and cross check the
website repo to sort it out

agramfort, 2016-07-02:
Is the pb on github.io or martinos.org ? If martinos I need to debug my rsync

jaeilepp, 2016-07-03:
I think [mne-tools.github.io](http://mne-tools.github.io/dev/auto_tutorials/plot_artifacts_correction_ica.html) is up to date. [Martinos](http://martinos.org/mne/dev/auto_tutorials/plot_artifacts_correction_ica.html) has not yet been updated.

agramfort, 2016-07-04:
rsync is telling me it's up to date with the master branch of github.io

jaeilepp, 2016-07-05:
It looks like it's only partially updated (as the title here suggests). See the tag at the bottom of the page for the ica tutorial (Last updated on 2016-07-01) and the index (Last updated on 2016-07-02).

agramfort, 2016-07-05:
should be fixed now.

jaeilepp, 2016-07-05:
Looks good. Closing.

closed 2016-07-05 by jaeilepp