mne-tools/mne-python #3385: .sfp import with long ch names

jona-sassenhagen, 2016-07-05:
The .sfp chan loc importer currently truncates long channel names. I came across this in a file with eye tracker channels included - 'L_GAZE_X' and 'L_GAZE_Y', which were truncated to have the same name, leading to all kinds of problems.

Could we simply do `dtype = np.dtype('S16, f8, f8, f8')` instead of the `dtype = np.dtype('S4, f8, f8, f8')` we currently have? 

This would still break at very long names. Possibly, a safer writer would process the strings line by line. But maybe long ch names are so rare it's not worth it.

jona-sassenhagen, 2016-07-05:
This possibly also applies to .elp, .hpts, .eloc/.loc.

Eric89GXL, 2016-07-05:
> Possibly, a safer writer would process the strings line by line. 

This seems like the safest solution to me. If we want to keep an upper limit we can use that of the max FIF channel name length (not sure what that is).

jona-sassenhagen, 2016-07-05:
Ok, I can do a string-based processor at least for .sfp (which I know), and possibly .loc. If I don't manage the others, I'll ping this thread again. Thanks guys.

jona-sassenhagen, 2016-07-05:
@teonbrooks maybe you wanna do this, otherwise I'll start doing something horrible with .split and stuff.

Eric89GXL, 2016-07-05:
@jona-sassenhagen we use `.split` elsewhere, it could be fine here

PR #3387 (2016-07-06): fix long sfp chan names
Replace file parser for .sfp files. Less efficient, but should not choke on long channel names.

~~Missing dedicated test for the specific scenario this is aimed to fix.~~

Closes #3385
closed 2016-07-06 by agramfort