mne-tools/mne-python #3393: List of evoked to plot_evoked

jaeilepp, 2016-07-06:
Just realized that we don't have a convenient function for plotting different conditions to same butterfly plot like this.
[image]

Should be quite straightforward by modifying plot_evoked.

agramfort, 2016-07-06:
plot_topo does it but for all channels in a topo. You can then click

jaeilepp, 2016-07-06:
Sure, I just feel that ``plot_evoked`` should also accept a list. We could also add some intelligence to it. Legends and such.

jona-sassenhagen, 2016-07-06:
Doesn't really work with butterfly plots, actually ...

But yes, this is a great anger to ERP people using MNE.

For anyone interested in implementing it, I'd suggest:
- look at ERPlab for style. E.g., axes through origin
- allow plotting dicts of evokes, where the key becomes the label
- think about somehow measuring uncertainty

>plot_topo does it but for all channels in a topo. You can then click

Doesn't work well for %matplotlib inline. And plot_topo needs a lot of love IMO ...

agramfort, 2016-07-06:
can you write a gist of the API / usage so we know what to be coded?
can you also share an image of how it should look like?
also some EEG friends want to swap y axis :)

jona-sassenhagen, 2016-07-06:
>also some EEG friends want to swap y axis :)

Argh yes.
EEGLAB uses a global param for this btw. (They default to positive up, but a lot of EEG people don't.)

jona-sassenhagen, 2016-07-06:
Typical ERP plots:
[image]

[image]

(Also often in too layout)

And an example of a single-ectrode condition plot I've recently done:
[p3cpz.pdf](https://github.com/mne-tools/mne-python/files/349509/p3cpz.pdf)
I'd add ticks though, dunno why I dropped them here.

agramfort, 2016-07-06:
shameless self promotion :)

what API do you envision? function would plot one channel at a time? the
average of a list of channels? accept axes params?

can you come up with a clean minimal API?

jona-sassenhagen, 2016-07-06:
>shameless self promotion :)

Actually that paper is reasonably well cited (for what it is) ... I just wanted people to know that it's *my* impression of what a decent plot looks like.

>what API do you envision?

Single-electrode plots only really make sense for contrasts, so I'm thinking of an mne.viz function rather than a method. It could be an extension of plot_topo, but for a function that focuses on single-electrode stuff, that doesn't seem like the right name. Maybe it would make sense to have a single-electrode function. (Then, plot_evoked_topo could be refactored to call this function for the axes. And it can easily plot into rectangular arrays.)

I'll think about it some more and ask the ERP people I know (speaking of, @teonbrooks ...). But for a sketch:

`mne.viz.plot_compare_evoked(list_or_dict_of_evoked, picks, ax, vline, ...)`

ERP people love different line styles, too, so maybe that could be accommodated for.

What would be really cool is if we could have something like

```
def plot_compare_evoked(evokeds, ..., picks, ...):
    evokeds: list of evoked | dict of evoked | list of lists of evokeds | dict of lists of evokeds
        ... if dict, plot dict keys as condition labels.
        ... if list of lists or dict of lists, plot the grand mean and 95%
        confidence interval over the evoked instances for each condition.
....
    picks: int | list of int
         What channel do plot. If multiple electrodes, their mean is plotted.
```

agramfort, 2016-07-06:
i like the API you suggest.

any volunteer?

jona-sassenhagen, 2016-07-06:
I'd actually like to do it - a lot more fun than the IO/montage fixes I *should* be doing.

jona-sassenhagen, 2016-07-06:
Unless @mmagnuski wants to do it.

agramfort, 2016-07-06:
> I'd actually like to do it - a lot more fun than the IO/montage fixes I
*should* be doing.

you can do both :)

mmagnuski, 2016-07-06:
Go ahead, take it, I'm pretty busy this week (that's why my PRs are so tiny :)). This would be really nice to have!

jona-sassenhagen, 2016-07-06:
http://nbviewer.jupyter.org/gist/jona-sassenhagen/cb5219cf65bf9a0dea2fce215478aaab

What do you think?

mmagnuski, 2016-07-07:
Really cool! I really like the way mapping colors/linestyles to conditions work

agramfort, 2016-07-07:
Looks great !

jona-sassenhagen, 2016-07-07:
What do you think of this API?

```
def plot_compare_evokeds(evokeds, picks=None, conditions=None, ch_names=None,
                        colors=None, linestyles=['-'], styles=None, vlines=[0],
                        errorbar=1.96, truncate_yaxis=True, ymin=None, ymax=None,
                        ax=None):
    """ Plot, for multiple conditions, individual evoked time courses of one or
    the mean of multiple sensors
    
    This function is useful for comparing ER[P/F]s at a specific location. It
    plots grand averages or, if supplied with a list/dict of lists of evoked
    instances, grand averages plus confidence intervals.
    
    Parameters
    ----------
    evokeds : instance of mne.Evoked | list | dict
        If a single evoked instance, it is plotted as a time series.
        If a dict, the contents are plotted as single time series each and the
        keys are used as condition labels.
        If a [dict/list] of lists, the unweighted mean is plotted as a time
        series and the parametric confidence interval is plotted as a shaded
        area. All instances must have the same shape - channel numbers, time
        points etc.
    picks : int | list of int
        The indices of the sensors to plot. Must all have the same channel
        type.
        If None, the global field power is plotted instead.
    conditions : list of str | dict | None
        A list of condition names. If `evokeds` is a list or dict, must be None
        or of the same type as `evokeds`. If both `evokeds` and `conditions`
        are dicts, must be a dict mapping the keys of `evokeds` to labels for
        the plot legend. If both are list or `evokeds` is a single Evoked
        instance, will be used for condition labels.
    ch_names : None
        A str or list of str for the title. If None, will be inferred from
        picks.
???     If picks is None, it will be inferred from ch_names; in this case, ch_names
???     must correspond to actual names of the evokeds instances.
    colors : list | dict | None
        If a list, will be sequentially used for line colors.
        If a dict, can map evoked keys or HED-like tags to conditions. For
        example, if `evokeds` is a dict with the keys "Aud/L", "Aud/R",
        "Vis/L", "Vis/R", `colors` can be dict(Aud='r', Vis='b') to map both
        Aud/L and Aud/R to the color red.
        If None, a sequence of desaturated colors is used.
    linestyles : list | dict
        If a list, will be sequentially used for evoked plot linestyles.
        If a dict, can map the `evoked` keys or HED-like tags to conditions.
        For example, if evokeds is a dict with the keys "Aud/L", "Aud/R",
        "Vis/L", "Vis/R", `linestyles` can be dict(L='--', R='-') to map both
        Aud/L and Vis/L to dashed lines.
    styles : dict | None
        If a dict, keys must map to evoked keys or conditions, and values must
        be a dict of legal inputs to `matplotlib.pyplot.plot`. These
        parameters will be passed to the line plot call of the corresponding
        condition, overriding defaults.
    vlines : list
        A list of integers corresponding to the positions, in milliseconds,
        at which to plot dashed vertical lines.
    errorbar : float | None
        If not None and `evokeds` is a [list/dict] of lists, a shaded
        uncertainty band is drawn around the mean and this sets the width of
        the band as (`*$errorbar` * standard error of the mean). I.e,
        if 1.96 (the default), the 95% parametric confidence interval is drawn.
        If None, no shaded error band is plotted.
    truncate_yaxis : bool
        If True, the left y axis is truncated to reduce visual clutter.
    ax : None | `matplotlib.pyplot.axes` instance
        What axes to plot to. If None, a new axes is created.
    ymin, ymax : float | None
        If not `None`, can be used to manually scale the y axis.
    
    Returns
    -------
    fig : Figure
        The figure on which the plot is drawn.
    """
```

jona-sassenhagen, 2016-07-07:
A few example plots ...

A call with no parameters (the example data): the GFP is shown, there is a nonsensical legend, and distinct colors.
`plot_compare_evokeds(list(evokeds.values()));`

[image]

Plotting 4 conditions, setting color and linestyle with HED tags, and manually setting condition names.

```
cond_names = {"Vis/R":"Visual Right",
              "Vis/L":"Visual Left",
              "Aud/R":"Auditory Right",
              "Aud/L":"Auditory Left"}
f = plot_compare_evokeds(evokeds, 55, conditions=cond_names,
                     colors={"Vis":'Crimson', 'Aud':'CornFlowerBlue'},
                     linestyles={"R":"-", "L":"--"});
```

[image]

Plotting a dict of lists of evokeds: the 95% parametric CI is shown. 4 Electrodes are selected and the mean is plotted.

```
plot_compare_evokeds(evokeds_2, picks=picks, colors={"tight":'Crimson', 'loose':'CornFlowerBlue'},
```

[image]

jona-sassenhagen, 2016-07-07:
What's currently missing is a bunch of input checks, tests, and dealing with MEG data. Otherwise I can PR.

agramfort, 2016-07-07:
the conditions and ch_names params seem redundant to me.

jona-sassenhagen, 2016-07-07:
`conditions` is important if you want to plot a list (not a dict), or if you want different names for the conditions than the dict keys - e.g., if you want a HED for the keys so you can map colors to tags, but want "normal" names for the legend. See example 2.
It would only be redundant if we don't allow list input.

`ch_names` helps if you don't want to call your channel name selection "EEG 50 + EEG 51 + EEG 52", but for example "Posterior electrodes". I was also thinking about allowing something like `plot_compare_evokeds(evokeds, ch_names=['Cz', 'CPz'])` and it would auto infer the picks from that.
It's arguably not that important, but IMO it's nicer to do 

`plot_compare_evokeds(evokeds, ch_names=['Cz', 'CPz'])`

than

```
picks = [evokeds[condition].ch_names.index(ch) for ch in ['Cz', 'CPz']]
plot_compare_evokeds(evokeds_2, picks=picks)
```

agramfort, 2016-07-07:
>
> conditions is important if you want to plot a list (not a dict),
>
pick one option and stick to it.

> or if you want different names for the conditions than the dict keys -
> e.g., if you want a HED for the keys so you can map colors to tags, but
> want "normal" names for the legend. See example 2.
> It would only be redundant if we don't allow list input.
>
first reaction : keep it simple stupid

make the API as simple as possible and I don't care if you need 1 extra
line to do exactly what you want.

> ch_names helps if you don't want to call your channel name selection "EEG
> 50 + EEG 51 + EEG 52", but for example "Posterior electrodes". I was also
> thinking about allowing something like [...]
you can always change the figure after it's rendered so again keep it
simple.

> plot_compare_evokeds(evokeds, ch_names=['Cz', 'CPz'])
>
> than
>
> picks = [evokeds[condition].ch_names.index(ch) for ch in ['Cz', 'CPz']]
> plot_compare_evokeds(evokeds_2, picks=picks)
>
>  I can live with this extra line or a call to pick_channels

jona-sassenhagen, 2016-07-07:
This addresses #1939 btw.

jona-sassenhagen, 2016-07-07:
I'm surprised you didn't object to `styles` ... I actually think it's less important than `ch_names` or `conditions`. What about dropping `styles`?

I guess a good side to dropping `conditions` is you have no plots with default labels anymore. So I think I can live with dropping it.

>you can always change the figure after it's rendered so again keep it simple.
Well yes, but this type of plot is probably a thing the typical beginner, who maybe doesn't know much python, has the most need for ...
And you will in the end have this for almost *every single call* to this function.
How about overloading pics to also take str | list of str, for auto inferring picks?

Also I take it the French team won that soccer game?

jona-sassenhagen, 2016-07-07:
This one is superficially similar to #3123. But I think there actually isn't much overlap.

In principle, one could consider factoring out the spines etc. here and re-using them for other time series plots, but the spines for GFP plots and regular ERPs are actually rather different.

Just a note.

jona-sassenhagen, 2016-07-07:
@wmvanvliet maybe you also have an opinion.

dengemann, 2016-07-07:
Something is not simple about this API I agree. Intuitively either list or dict ...

jona-sassenhagen, 2016-07-07:
dict is much better.

dengemann, 2016-07-07:
I also like list, I simply don't get my head around how `conditions` is intended to be used. Some further simplification and pruning would be good. Impose a few strong usecases, the rest is for matplotlib cracks.

jona-sassenhagen, 2016-07-07:
Well if evokeds is a dict, my idea was you could do something like
```
cond_names = {"Vis/R":"Visual Stimulus, Right Visual Field",
              "Vis/L":"Visual Stimulus, Left Visual Field",
              "Aud/R":"Auditory Stimulus, Right Ear",
              "Aud/L":"Auditory Stimulus, Left Ear"}
f = plot_compare_evokeds(evokeds, 55, conditions=cond_names,
                     colors={"Vis":'Crimson', 'Aud':'CornFlowerBlue'},
                     linestyles={"R":"-", "L":"--"});

And it would be exactly as if you skipped conditions, but the legend would be different.

The other use case for dicts would be enforcing a certain ordering of the conditions - i.e., if you want the 2 Viz first and then the two Aud, you'd give it a list like that.

It'd be confusing because these two use cases are mutually incompatible, but share one kwarg.

The third would be if you give it a list, cause you still need condition labels. But I think for various reasons, dict-only input is much better than list-only input.

So that's my vote: no support for lists and no `conditions` param.

Eric89GXL, 2016-07-07:
Evoked already have a comment field that should ideally be usable for
setting titles and such. So by only using dict you would force people into
doing redundant things (mapping evoked.comment to evoked) sometimes, right?

It is less work to pass a list than a dict. If people want different titles
than they can modify evoked.comment, no?

agramfort, 2016-07-08:
I like the idea of using the comment attribute.

wmvanvliet, 2016-07-08:
About the 'typical ERP plots' you mentioned above: thats why we have the `make_grid_layout` function.

I agree that `plot_evoked_topo` needs a lot of love; you made a valiant effect in the past which unfortunately stranded on architectural design.

I don't see the use case for plotting the mean of several channels. I never encounter this in paper. At most, I sometimes encounter a topo plot where people have averaged across channels in order to make 'frontal', 'temporal', 'occipital', etc. selections. This functionality has been discussed some time ago I think, but was never really picked up.

Wow, that example plot you made looks gorgeous! What a great idea to uncouple the ticks from the axes that go through the origin, so the lines don't overlap with the tick labels. Yes, I'll have some of that please!

This function should be used by `plot_evoked_topo` to plot single channels when clicked on. The user should also be able to call it manually to quickly plot a single channel.

Right now, the function tries to do too much. The API can be simplified IMO:

    evokeds : instance of mne.Evoked | list | dict

There's no need for a `dict`. `Evoked`'s have a `.comment` field for exactly this reason. If the user really wants to specify custom labels for the legend, let him make a call to `plt.legend(['what', 'ever', 'you', 'want'])` and the legend will be overwritten.

    picks : int | list of int

From a users point of view, it would be more convenient if this were `ch_name` and a string. I don't see the point of allowing a list here and averaging across the channels... when is that ever helpful? It should be a required parameter: we already have a way to plot GFP (`evoked.plot`); comparing conditions by comparing GFP is out of scope for this function.

    ch_names : None
        A str or list of str for the title. If None, will be inferred from
        picks.

Just call it `title` and let it be a string. By default it will be the channel name.

    vlines : list

Drop this one, it's easier for a user to just call `plt.axvline`.

    errorbar : float | None
        If not None and `evokeds` is a [list/dict] of lists, a shaded
        uncertainty band is drawn around the mean and this sets the width of
        the band as (`*$errorbar` * standard error of the mean). I.e,
        if 1.96 (the default), the 95% parametric confidence interval is drawn.
        If None, no shaded error band is plotted.

It's not an errorbar so lets rename this parameter. How about `show_conf_int`? Why specify it as multiple of the standard error of the mean? It's much more straightforward to specify it as percentage, i.e. `.95` for the 95% parameteric confidence interval.

Finally, I don't like the name `plot_compare_evokeds`. How about `plot_evoked_channel` and have a method `Evoked.plot_channel`?

jaeilepp, 2016-07-08:
I basically agree with @wmvanvliet , except for
> Finally, I don't like the name plot_compare_evokeds. How about plot_evoked_channel and have a method Evoked.plot_channel?

I personally would like to see ``plot_compare_evokeds`` where you could pass a list of evoked and ``evoked.plot_compare`` where you could pass the evoked (or a list) to compare the two.

agramfort, 2016-07-08:
+1 for the function. No strong feeling about the word "compare"

wmvanvliet, 2016-07-08:
`plot_evoked_topo` also compares. And this function would only compare if you gave a list of evokeds.

jona-sassenhagen, 2016-07-08:
Okay that's a lot of talk, thanks guys. It seems there is a strong consensus for lists, but I'll at least once try making the case for dict.

**I don't think evoked.comments for conditions is good. I think it's bad.**

evoked.comments not used much for condition names in the code: there is one example (plot_topo_compare_conditions.py), and one function (plot_snr_estimate). In the new tutorials, dict keys are used for cond names instead.

I also don't think it's a good place to put a condition. It's *comment*, not condition, after all; under comment, I expect something like "bad subject" or a detailed explanation, not a condition label like "Visual/Left".

I also don't think lists plus the comments field is a good way to access items. Contrast (from the tutorials):

```
all_evokeds = dict((cond, epochs[cond].average()) for cond in event_id)
print(all_evokeds['left/auditory'])

for cond in all_evokeds:
    all_evokeds[cond].plot_joint(title=cond)

# or
for cond, evoked in all_evokeds.items():
    evoked.plot_joint(title=cond)

```

That's simple, and pythonic.
Now a list-and-comments way:

```
conditions = list(event_id.keys()
all_evokeds = [epochs[cond].average()) for cond in conditions]
print(all_evokeds[conditions.index('left/auditory')])

# or worse
idx = [ii for ii, evoked in enumerate(all_evokeds) if evoked.comments == 'left/auditory']
print(all_evokeds[ii])

for cond, evoked in zip(all_evokeds):
    evoked.plot_joint(title=cond)
```

The dict way is also much like how we select conditions from epochs. We do `epochs["left/auditory"]` after all.

So I think just from a code cleanliness perspective, using good old Python dicts is the clearly superior way of handling evoked objects. 

Beyond that, **MNE functions don't actually support comments for conditions.**

The comments field will be set to completely the wrong thing for grand averages. And we're talking here about a function that is probably *mostly* going to be used for grand averages!
https://github.com/mne-tools/mne-python/blob/6be10dbd4462874acabad42eab48a277754ff6f0/mne/utils.py#L2136

`grand_average.comment = "Grand average (n = %d)" % grand_average.nave`

So for grand averages, the comment field does *not* hold the condition.
It can't, either; because it's clearly possible to do grand averages of evokeds with all kinds of comments, and I don't think it's a good idea to try and parse/magic our way through this.
The comment field of GAs *does* a good job of informing users about important stuff: that this is a grand average.

In fact, if you want to plot a difference waves, it's even worse. `combine_evoked` simply uses the first evoked's comment, whatever you do. So if you do "Aud/L" - "Aud/R", the comment will be "Aud/L".
For `combine_evoked`, it's in principle possible to construct a comment string based on combining the weights with the individual comments, but if you think of some complex example (a combination of addition and subtraction of 8 conditions, for example), it looks silly.

So for difference waves, the comment field does *not* hold the condition either.
And this is not documented anywhere either, meaning users either did not expect GA or difference comment fields to hold the condition anyways and never checked, or they know it doesn't do that, or they manually set it.
All in all, it seems to me for the stuff you actually WANT to plot an ERP for - grand averages, and condition differences - the code actively fights against using `comments` to store conditions.

So for API for this function here, you can either have dicts like

```
evokeds = {condition:grand_average([epoch_[condition].average()
                                   for epoch_ in all_epochs])
                                  for condition in event_id}
plot_compare_evoked(evokeds)
```

or lists like

```
# with a conditions param
conditions = list(event_id.keys())
evokeds = [grand_average([epoch_[condition]
                          for epoch_ in all_epochs])
           for condition in conditions]
plot_compare_evokeds(evokeds, conditions)
```

or

```
# with manually setting comments
conditions = list(event_id.keys())
evokeds = [grand_average([epoch_[condition]
                          for epoch_ in all_epochs])
           for condition in conditions]
for condition, evoked in zip(conditions, evokeds):
    evoked.comments = condition
plot_compare_evokeds(evokeds)
```

IMO no. 1 is clearly the best.

`linear_regression_raw` already returns a dict of evokeds. The alternatives are clearly inferior (returning an evoked and a condition list, or expecting the user to rely on comments).

**In summary:**
While the docs claim that `evoked.comments` will "usually" hold the condition, I think 1. that's semantically surprising, 2. unhelpful API, 3. not supported by the actual code.

In contrast, dicts 1. are  semantically clear (keys are labels!), 2. keep symmetry between Epochs and Evoked, 3. standard Python.

So I think the standard way of dealing with multiple Evokeds of different conditions, and what the docs and examples should (continue to) recommend, and what new API might consider beginning to support, should be dicts.

jona-sassenhagen, 2016-07-08:
I don't have a strong intuition about how to call the function - plot_compare_evoked, plot_evoked_channel both seem fine to me.

>I agree that plot_evoked_topo needs a lot of love; you made a valiant effect in the past which unfortunately stranded on architectural design.

On me being an even worse coder back then.

>This function should be used by plot_evoked_topo to plot single channels when clicked on. The user should also be able to call it manually to quickly plot a single channel.

Long term, it might become the function plot_evoked_topo uses internally to generate individual plots, too. You'd just have to drop the left and bottom spines, and place them elsewhere.

>I don't see the use case for plotting the mean of several channels. 

I fear that's fairly common: whenever you do a ROI. It'd also be there for lasso selection over plot_evoked_topo. Also, picks usually allows a list.

>From a users point of view, it would be more convenient if this were ch_name and a string

Yes, but in similar functions (e.g. plot_epochs_image), we use picks.

>we already have a way to plot GFP (evoked.plot); comparing conditions by comparing GFP is out of scope for this function.

On one hand, I think GFP comparisons is nice to have, on the other hand, the code isn't the easiest and it fucks with the CI. Any other opinions?

>Just call it title and let it be a string. By default it will be the channel name.

Yes, that's probably for the best.

>Drop this one, it's easier for a user to just call plt.axvline.

Only if you're okay with the vline not matching the left spine in height. Having the vline match the y axis in height is not trivial (unless you want to do it manually).

>It's not an errorbar so lets rename this parameter. How about show_conf_int? Why specify it as multiple of the standard error of the mean? It's much more straightforward to specify it as percentage, i.e. .95 for the 95% parameteric confidence interval.

I think seaborn just calls it CI, and you feed it a fraction. I'll do it that way.
Todo list: bootstrapped CIs.

mmagnuski, 2016-07-08:
I wanted to comment on the usefulness of averaging channels in this function - I think it is useful and I do it often.
Especially if I am using cluster correction and then want to show agreggated cluster effect for different levels of some predictor - I then average channels that make up most of the cluster mass for example.

I like overloading picks with string / list of strings. But this is probably a good predictor of core devs not liking it. :)
Maybe `ci` instead of `errorbar` (along the suggestion of @wmvanvliet)?
I also agree it is easy enough to call `plt.legend` or `plt.title`so that some arguments are not needed.

jona-sassenhagen, 2016-07-08:
>Maybe ci instead of errorbar (along the suggestion of @wmvanvliet)?

>I also agree it is easy enough to call plt.legend or plt.titleso that some arguments are not needed.

Yeah, I think these two are obvious.

mmagnuski, 2016-07-08:
Also: +1 on not using `.comment`. I like passing dict (and possibly like passing dict or list even more)

wmvanvliet, 2016-07-08:
@jona-sassenhagen ok, you convinced me that using `.comment` is a bad idea. Dicts it is. And it didn't occur to me that the vline height would not match, so you are right about that parameter as well.

Eric89GXL, 2016-07-08:
I'm okay with not having to use `.comment`. How often do you expect users to use this function where the `.comment` is actually correct? Because for them, having to pass a dict instead of a list is cumbersome and redundant. If it's not going to be that often, that's fine, but if e.g. comparing `Aud/L` and `Aud/R` conditions from a single user is an expected use case, then it seems like the list solution is better because if the list is empty, it can just use the `.comment` field.

jona-sassenhagen, 2016-07-08:
I personally would expect that the more important use case - the one where people actually need pretty defaults - is grand averages. What do you need single-subject plots for? Usually just checking if there is some crazy artifact or whatever, for which you don't really need this.

But maybe that's cause I'm coming from a language-ERP/EEG background, where individual subjects are really noisy and you can only really see stuff in grands and diffs.

jona-sassenhagen, 2016-07-09:
Any further comments @agramfort @jaeilepp @dengemann ?

agramfort, 2016-07-09:
ok point taken

I start to be convinced that dict input is useful and I now think that list
input is also useful to have a dummy API. Regarding the picks/ch_names I
would just say: choose the param that you think is the most useful picks or
ch_names.

please open the PR when you have time.

jona-sassenhagen, 2016-08-30:
Closed by #3413

closed 2016-08-31 by teonbrooks

psychoxiatao, 2018-12-07:
> A few example plots ...
> 
> A call with no parameters (the example data): the GFP is shown, there is a nonsensical legend, and distinct colors.
> `plot_compare_evokeds(list(evokeds.values()));`
> 
> [image]
> 
> Plotting 4 conditions, setting color and linestyle with HED tags, and manually setting condition names.
> 
> ```
> cond_names = {"Vis/R":"Visual Right",
>               "Vis/L":"Visual Left",
>               "Aud/R":"Auditory Right",
>               "Aud/L":"Auditory Left"}
> f = plot_compare_evokeds(evokeds, 55, conditions=cond_names,
>                      colors={"Vis":'Crimson', 'Aud':'CornFlowerBlue'},
>                      linestyles={"R":"-", "L":"--"});
> ```
> [image]
> 
> Plotting a dict of lists of evokeds: the 95% parametric CI is shown. 4 Electrodes are selected and the mean is plotted.
> 
> ```
> picks = [evokeds["tight"].ch_names.index(ch) for ch in ["CPz", "Pz", "P3", "P4"]]
> plot_compare_evokeds(evokeds_2, picks=picks,
>                      colors={"tight":'Crimson', 'loose':'CornFlowerBlue'},
> ```
> [image]

Sorry, why I can not plot the CI (shadow) with this code? Thanks.

mmagnuski, 2018-12-07:
@psychoxiatao You have to be more specific, what code and data are you using? Remember that passing a single ERP for each condition would not give you CI, you need to use a list (or dict of lists) of ERPs for each condition.

psychoxiatao, 2018-12-07:
> @psychoxiatao You have to be more specific, what code and data are you using? Remember that passing a single ERP for each condition would not give you CI, you need to use a list (or dict of lists) of ERPs for each condition.

Thanks for your reply. How can I create a list of ERPs for the different individual? How can I combine each individual's evoked data toghter?

mmagnuski, 2018-12-07:
You create the list similarily to any other python list, you just append ERPs to it, for example:
```python
import os
import mne

files = os.listdir('C:/data/experiment/fif_files')

erp_list = list()
condition = 'RT < 0.5'

for fname in files:
    epochs = mne.read_epochs(fname)
    erp_list.append(epochs[condition].average())
```