mne-tools/mne-python #3401: ENH: info shows approx. mem size?

jona-sassenhagen, 2016-07-07:
When you call a Pandas Dataframe's info method, it shows you an approximation of the memory usage:

```
<class 'pandas.core.frame.DataFrame'>
MultiIndex: 17182 entries, (Vis/L, 0, -200) to (Vis/R, 241, 500)
Data columns (total 59 columns):
EEG 001    17182 non-null float64
EEG 002    17182 non-null float64
...
EEG 060    17182 non-null float64
dtypes: float64(59)
memory usage: 7.9+ MB
```

Would it in principle be possible to show something like this for MNE objects, for example, in the __repr__, the info's __repr__, or as a special private method?

Not a big thing, but might be cool for when you want to parallelize across data sets, or want to estimate if an operation will blow up your memoy.

Eric89GXL, 2016-07-07:
That would be cool

mmagnuski, 2016-07-07:
+1 from me too

Eric89GXL, 2016-07-07:
You could probably get pretty far by basing the estimate off the logic used
by object_hash.

But that might be too slow. If it is, we can get a rougher estimate from
just don't .data or ._data, as the object dictates.

Are you on it? If not I can try soon.

jona-sassenhagen, 2016-07-08:
>Are you on it? If not I can try soon.

I don't know how I'd even approach this :) And I have a huge backlog of other stuff (EEGLAB IO fixes, ERP plot, my open PRs etc).

jona-sassenhagen, 2016-07-10:
Closed by #3408

PR #3408 (2016-07-11): MRG: Add memory usage to primary objects
Closes #3401.

For example:
```Python
>>> print(mne.io.read_raw_fif('sample_audvis_raw.fif'))
<Raw  |  sample_audvis_raw.fif, n_channels x n_times : 376 x 166800 (277.7 sec), ~3.8 MB>
>>> print(mne.io.read_raw_fif('sample_audvis_raw.fif', preload=True))
<Raw  |  sample_audvis_raw.fif, n_channels x n_times : 376 x 166800 (277.7 sec), ~482.3 MB>
>>> print(mne.read_epochs('sample-epo.fif', preload=False))
<EpochsFIF  |  n_events : 121 (all good), tmin : -0.0998976065792 (s), tmax : 1.0006410259 (s), baseline : (-0.0998976081609726, 0.0), ~3.8 MB,
 u'Audito [... cut]
closed 2016-07-10 by jona-sassenhagen