mne-tools/mne-python #3411: legend missing with evoked.plot_joint() and evoked.plot()

jasmainak, 2016-07-09:
I am trying to plot evokeds using the super-cool spatial colors that Jona added. However, when the colors are not inferred from the layout, there is no legend. I think there should be a legend whenever there are colors. To reproduce, look at evoked from HCP ...
cc @jona-sassenhagen 
[image]

agramfort, 2016-07-09:
can you share a -ave.fif file to replicate?

jasmainak, 2016-07-09:
sure, here you go: https://www.dropbox.com/s/hwylpk72xfq1g6x/test.fif?dl=0

agramfort, 2016-07-10:
the pb comes from the fact that the magnesWH3600 layout has channels MEG001
etc. while your file uses the native BTi ch_names A22 etc. The cleanest
option would be to have new layout magnesWH3600A.lout for ch_names starting
with A and not MEG.

jasmainak, 2016-07-10:
okay, but I think there's also a problem with the logic. It will recur when there's another file with missing layout if we don't fix the logic. @jona-sassenhagen do you have time to dig into it? Otherwise, I'll take a look in the coming days

agramfort, 2016-07-10:
see my attempt:

https://github.com/agramfort/mne-python/commit/1de18e27ab4aa39b614090c30c9f980c05b0f4c1

however the layout.names = ch_names is not good enough. The order of
channels don't match :(

jona-sassenhagen, 2016-07-10:
I have a long backlog, would appreciate if somebody else could do this :(

dengemann, 2016-07-11:
>option would be to have new layout magnesWH3600A.lout for ch_names starting
with A and not MEG.

that would be the solution, I confirm this. It was what I had already discussed with @jona-sassenhagen a months ago. It is in fact an mne-python problem, not an mne-hcp problem.

jasmainak, 2016-07-11:
actually with @agramfort we realized that the spatial colors _always_ use the location in `info` to infer the color. So, the layout won't fix the problem.

Therefore, the solution seems to be to have an `apply_trans(info)` method which converts the `info` into another coordinate system. For now, I'll just fix it for my processing pipeline. But it appears that a more permanent solution is to have something to map the infos back and forth between the two coordinate systems.

dengemann, 2016-07-11:
yes that was what I was suggesting in the chat; since bti native uses millimeter you can guess evrything

closed 2016-07-11 by agramfort