mne-tools/mne-python #3416: [BUG] joint_plot does not take exclude param

jasmainak, 2016-07-11:
I tried the following:

```py
from mne.datasets import sample
from mne import read_evokeds

path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'

condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))

times = 0.1
evoked.plot_joint(times, ts_args=dict(exclude=[]))
```

and it didn't work ...

agramfort, 2016-07-11:
are you sure that exclude is a valid parameter for ts plot?

jona-sassenhagen, 2016-07-11:
As `exclude` applies to both the topo plot and the time series plot, it is a direct arg to plot_joint (see the doc).

We don't explicitly say this in the doc, but if it's already a param to plot_joint, it can't be passed to one of the two underlying functions.

(ie., the correct solution is `evoked.plot_joint(times, exclude=[])`)

jasmainak, 2016-07-11:
okay, I hadn't noticed that there was already an `exclude` parameter (hemineglect :).

so, how many params are actually customizable then? It seems to me not many. In that case, why kwargs? In any case, it would be nice to document it explicitly saying what's allowed rather than focusing on what's not allowed (if there aren't very many ...)

jona-sassenhagen, 2016-07-12:
You can pass approximately 30 args via the two kwarg params. Check the code or the doc; as the doc says, most of what you insert there is passed directly to evoked.plot or evoked.plot_topomap.

jasmainak, 2016-07-12:
okay got it. Maybe a more informative error message + better docs would do it for me.

agramfort, 2016-07-12:
PR welcome @jasmainak 

closing

closed 2016-07-12 by agramfort