mne-tools/mne-python #3417: `$ mne make_scalp_surfaces` requires access to screen

christianbrodbeck, 2016-07-12:
For me, `$ mne make_scalp_surfaces` ends with

```
...
3. Creating medium tessellation...
3.1 Decimating the dense tessellation...
This program needs access to the screen.
Please run with a Framework build of python, and only when you are
logged in on the main display of your Mac.
```
(OS X, Anacond 2.3, on master as well as 0.12). The cause seems to be `mne.decimate_surface` which calls on TVTK. Is this a known issue? The normal solution to this error would be to run a script with `pythonw` (the framework build) instead of `python` but I can't figure out how the `$ mne` command can be invoked with `pythonw`?

agramfort, 2016-07-12:
you mean it fails without X server?

christianbrodbeck, 2016-07-12:
Not sure what you mean, I am just running the command in the terminal. Should I run the command in a special way or install something? I have XQuartz installed if that is it.

agramfort, 2016-07-12:
weird. tvtk is a dependency of mayavi so it should be lighter and work if
mayavi works on your system. Sorry I can't tell more

Eric89GXL, 2016-07-12:
`wx` complains about framework builds like this. Try doing `ETS_TOOLKIT=qt4 command`

christianbrodbeck, 2016-07-13:
Thanks @Eric89GXL that was it! Would you be opposed to adding `os.environ['ETS_TOOLKIT'] = 'qt4'` to the `$ mne make_scalp_surfaces` command as we do for the `$ mne coreg` command?

Eric89GXL, 2016-07-13:
I don't think we should really add that, if someone wants to or needs to
use wx then they can't. It would be better to catch the error (I think it's
actually SystemExit) and say something informative

christianbrodbeck, 2016-07-13:
So would it also be preferable to handle the environment for the GUI that way?

Eric89GXL, 2016-07-13:
Yeah I think so

christianbrodbeck, 2016-07-13:
addressed in #3422

closed 2016-07-13 by christianbrodbeck