mne-tools/mne-python #3420: Bug on import of EDF files with accentuated annotations

deep-introspection, 2016-07-13:
Hey!
So importing an EDF file, I have a crash
> ValueError: byte must be in range(0, 256)

occuring at this level: https://github.com/mne-tools/mne-python/blob/f04d1460d10ebe3b9466ccad9c077040d42a6d5f/mne/io/edf/edf.py#L258

Digging into the code, I realize that the (French) people who have recorded the EEG may have put characters with accents.

By changing the above part of the code to :
```
try:
    tals.extend([i % 256, i // 256])
except:
    print i, tals[10:].decode('ascii', 'ignore')
    raise('DEBUG')
```
I obtained this:
> -5779 0000+44+1166PHOTO 0.5Hz+45+1172+1172.200000+46+1172Yeux ouverts+47+1178+1178.840000+48+1178Yeux fer

It's indeed French, and the error occurs exactly at the level of the annotation **Yeux fer** **_mé_** (if you take i = -5779, the i % 256 gives '**m**', so I assume the original second character causing the bug was '**é**'...

So, what should be the best way to handle this?

Thanks!
Guillaume

dengemann, 2016-07-13:
cc @teonbrooks @jaeilepp

jaeilepp, 2016-07-13:
@deep-introspection can you try if [this](https://github.com/mne-tools/mne-python/pull/3421) fix works for you.

deep-introspection, 2016-07-13:
@jaeilepp Sorry but apparently not ... The question in first place is what does a negative **_i_** mean? Shall we just make a warning and replace the character with a neutral one?

Eric89GXL, 2016-07-13:
> Shall we just make a warning and replace the character with a neutral one?

It would be better to properly handle UTF-8 encoding. We do it everywhere else (that we know of), and if the file format allows it, there should be some reasonable way to do it.

jaeilepp, 2016-07-13:
Is the error message still the same on https://github.com/mne-tools/mne-python/pull/3421? I can't be sure of the problem without testing it with some data. Can you share the problematic file?

deep-introspection, 2016-07-13:
Yes, the error message is the same:
> ValueError: byte must be in range(0, 256)

The problematic file is [there](http://www.morphomemetic.org/ftp/test_eeg.edf).

deep-introspection, 2016-07-18:
@jaeilepp Can I help in any way?

jaeilepp, 2016-07-18:
I was able to get through the first error by basically skipping the problematic characters with the file you shared, but after that I encountered an error with overlapping events. I'm not too familiar with EDF data and not sure why that happens. Any ideas @teonbrooks @Eric89GXL ?

Eric89GXL, 2016-07-18:
> I was able to get through the first error by basically skipping the problematic characters with the file you shared

Not a great permanent solution...

> after that I encountered an error with overlapping events.

I don't really work with EDF so I can't help with the overlapping events stuff. Does it just mean that triggers we try to construct overlap? If so, the MNE-like thing to do would be to add them.

teonbrooks, 2016-07-18:
i don't understand the overlapping problem. could you explain it a little more?

Eric89GXL, 2016-07-18:
I assume he's talking about [this] (https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L207), which I assume is there because we haven't worked out how to deal with the overlaps. It might not always be possible to do so without having the user provide a new mapping. For example, if there is an event of type 1 and an event of type 3, we cannot simply add them (which is what I suggested above). But if they have no bitwise overlap, then we could without losing any information.

So I think @jaeilepp can't easily debug the current issue without dealing with the events issue, since his test file has the events issue. But for now @jaeilepp you can just uncomment that check during your testing, and obviously the events channel will be messed up, but you can deal with the encoding issue properly.

deep-introspection, 2016-07-19:
@jaeilepp, how did you skip properly the problematic characters?

I tried this:
```
try:
    tals.extend([i % 256, i // 256])
except:
    warn('Characters skipped.')
```

but this crash later on:

```
Setting channel info structure...
Creating Raw.info structure...
Reading 0 ... 1297999  =      0.000 ...  2595.998 secs...
edf2mne.py:19: RuntimeWarning: Characters skipped.
  preload=True, verbose=True)
Characters skipped.
Characters skipped.
Characters skipped.
Characters skipped.
Characters skipped.
Characters skipped.
Characters skipped.
Characters skipped.
Characters skipped.
Traceback (most recent call last):
  File "edf2mne.py", line 19, in <module>
    preload=True, verbose=True)
  File "~/github/mne-python/mne/io/edf/edf.py", line 564, in read_raw_edf
    preload=preload, verbose=verbose)
  File "<string>", line 2, in __init__
  File "~/github/mne-python/mne/utils.py", line 666, in verbose
    return function(*args, **kwargs)
  File "~/github/mne-python/mne/io/edf/edf.py", line 90, in __init__
    verbose=verbose)
  File "<string>", line 2, in __init__
  File "~/github/mne-python/mne/utils.py", line 666, in verbose
    return function(*args, **kwargs)
  File "~/github/mne-python/mne/io/base.py", line 342, in __init__
    self._preload_data(preload)
  File "<string>", line 2, in _preload_data
  File "~/github/mne-python/mne/utils.py", line 666, in verbose
    return function(*args, **kwargs)
  File "~/github/mne-python/mne/io/base.py", line 543, in _preload_data
    self._data = self._read_segment(data_buffer=data_buffer)
  File "~/github/mne-python/mne/io/base.py", line 442, in _read_segment
    cals, mult)
  File "<string>", line 2, in _read_segment_file
  File "~/github/mne-python/mne/utils.py", line 666, in verbose
    return function(*args, **kwargs)
  File "~/github/mne-python/mne/io/edf/edf.py", line 193, in _read_segment_file
    evts = _parse_tal_channel(data[tal_channel_idx])
  File "~/github/mne-python/mne/io/edf/edf.py", line 264, in _parse_tal_channel
    tal_list = re.findall(regex_tal, tals.decode(encoding='UTF-8'))
  File "/Users/XXX/anaconda/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe9 in position 10022: invalid continuation byte
```

jaeilepp, 2016-07-20:
I don't have the code with me now, but I remember it being quite hackish. I think I checked the range in two places, so probably a check before line 16 in edf.py will do it.

deep-introspection, 2016-07-20:
@jaeilepp Curious to see what hack you found. Line 16 don't make sense to me.

deep-introspection, 2016-07-26:
@jaeilepp ?

deep-introspection, 2016-07-26:
Ok, so I solved the problem by checking the validity of the characters:
```
for s in tal_channel_data:
    i = int(s)
    i0, i1 = i % 256, i // 256
    if  0 <= i0 < 128 and 0 <= i1 < 128:
        tals.extend([i0, i1])
    else:
        warn('Skipped invalid caracters...')
```
Then, for the overlapping events, I changed the _Raise_ of _NotImplementedError_ by a _Warning_.
The events are still available with:
```
raw = io.read_raw_edf('EDF/test_eeg.edf', preload=True)
raw._raw_extras[0]['events']
```
Shall I attempt a pull request?

agramfort, 2016-07-26:
+1 for PR

deep-introspection, 2016-07-26:
Done :) https://github.com/mne-tools/mne-python/pull/3457

jaeilepp, 2016-08-29:
@deep-introspection can this be closed now?

agramfort, 2016-08-29:
I think so.

closed 2016-08-29 by agramfort