mne-tools/mne-python #3424: BUG: time domain of the plot_pattern exceed range

teonbrooks, 2016-07-15:
Using the linear_model.plot_patterns, there is an error in the plot where the time points do not correspond to the epochs time range. We can also see this in the [example](http://martinos.org/mne/dev/auto_examples/decoding/plot_linear_model_patterns.html#sphx-glr-auto-examples-decoding-plot-linear-model-patterns-py) where the plot pattern should be from -200 to 500 ms but it exceeds it in the plot. this is similar to #3180.

cc: @jaeilepp, @EstiBlancoElorrieta

kingjr, 2016-07-17:
IMO this `LinearModel` only contains 1 line of useful code (i.e.`np.dot(X.T, np.dot(X, self.model.coef_.T))`) out of 650 which are just copying sklearn objects, and copying plotting documentation.

IMO we should remove the object and update the example to make the Haufe trick explicit.

Alternatively, we can transform it into a Mixin class that we apply to decoding objects that are compatible with this option (CSP? Xdawn? GAT? anything else?). I suggest we keep this Mixin class private for now.

cc @kaichogami @dengemann

jona-sassenhagen, 2016-07-17:
>Alternatively, we can transform it into a Mixin class that we apply to decoding objects that are compatible with this option (CSP? Xdawn? GAT? anything else?). I suggest we keep this Mixin class private for now.

Would be cool.

People ask me about straight-forward viz of GAT patterns a lot.

teonbrooks, 2016-07-18:
a mixin class with a `plot_pattern` method seems to be the best approach. i agree, i don't see a lot of the utility in the `LinearModel` class but maybe I'm missing something.

agramfort, 2016-07-18:
+1 for mixin as it will force consistent API.

Yes it's 2 lines of numeric code but we want to keep examples simple.

Remember when you started with python a few years ago. An example with 100
lines of code with numpy tricks and matplotlib is a bit frighting when you
start

dengemann, 2016-07-18:
I'd be ok with a mixin. Some doc would be good. Main problem to me is that
the haufe trick creates marginal stats, so it is not a simply a clean way
to show the weights but something different. You should look at both.

kingjr, 2016-07-18:
Ok we could do something like
```
class _SpatialFilter(BaseEstimator, TransformerMixin):
    def _filter_pattern(self, X, y):
         if hasattr(self.estimator, 'coef'):
            self.filters_ = self.estimator.coef.T
            self.patterns_ = np.dot(X.T, np.dot(X, self.filters_))
 ```

and call `self._filter_pattern` after fit for Xdawn, CSP & GAT. 

We could also go a step further to retrieve linear pipelines: e.g. 
```
clf = make_pipeline(StandardScaler(), PCA(), GAT())

class _SpatialFilter(BaseEstimator, TransformerMixin):
    def _filter_pattern(self, X, y):
         if isinstance(self.estimator, Pipeline):
            if hasattr(self.estimator.steps[-1][-1], 'coef'):
                 coef  = self.estimator.steps[-1][-1].coef
                 try:
                      self.filters_ = self.estimator.inverse_transform([coef])
                except Exception:
                      return
                self.patterns_ = ....
```

But maybe that takes us too far?

teonbrooks, 2016-07-18:
> Remember when you started with python a few years ago. An example with 100
> lines of code with numpy tricks and matplotlib is a bit frighting when you
> start

oh, sorry, I should clarify, I meant if we moved its plotting functions to a Mixin class, there wouldn't be much left. but I just realized that this class is also the generic stand-in for all the linear models. 

btw, I still do get a lil' frighten with all the np and mpl tricks. for example, our spatio-temporal cluster plot example is not for the faint of heart ;)

agramfort, 2016-07-18:
@kingjr I would say why not but maybe not a priority. I would keep this for
a sprint...

what you see in
http://martinos.org/mne/stable/auto_examples/decoding/plot_linear_model_patterns.html
took some time to get during the last sprint in paris.

teonbrooks, 2016-07-18:
in the meanwhile, does it anyone see the problematic lines for the time bound breach? sorry, I unfortunately don't have the time to search for it; just relaying the bug

agramfort, 2016-07-19:
it's because the plot patterns uses indices and does not know about tmin
AFAIK. If you set tmin=0 by cropping I think you can go around the pb. It's
indeed a bug though

kingjr, 2016-07-29:
After discussion with @agramfort and @dengemann we decided that the mixin class would not support plotting methods, because it would require passing measurement info to the object which does not comply to our decoding API.

Instead we suggest to do add to the future examples:
```
EvokedArray(object.patterns_, epochs.info).plot()
```

Eric89GXL, 2016-09-08:
@kingjr do you have time to fix this? Would be good to have for 0.13 which we want to release in the next couple of weeks.

kingjr, 2016-09-08:
most likely no

Eric89GXL, 2016-09-08:
I'll remove the milestone for 0.13 then unless someone else finds time

PR #3949 (2017-02-21): MRG+1: get patterns coefs with example
After a couple of iterations, I believe the best way to deal with the extraction and plotting of spatial filters and spatial patterns is to i) ensure that we use the LinearModel class at the end of our pipeline and ii) use a function that can extract and inverse transform these coefficients.

This API allows a full sklearn compatibility, and avoid adding additional complex attributes to `SearchLight` and `GeneralizationLight` (filters_, inverse_filters, inverse_patterns etc)

```python
clf = make_pipeline(Vectorizer(), StandardScaler(), LinearModel(Logist [... cut]
closed 2017-02-21 by kingjr