mne-tools/mne-python #3427: ENH: EMS with sklearn object?

kingjr, 2016-07-18:
@kaichogami there is a function `mne.decoding.ems.compute_ems` which could be implemented as an sklearn object to get a simple API.

Although the function looks awfully simple (it's a normalized differential response? Or I'm missing something). 

@dengemann Do you have an opinion on this?

agramfort, 2016-07-18:
+1 for sklearn API object for EMS

dengemann, 2016-07-18:
Would be neat to have such an object. But priority is low at this point

jona-sassenhagen, 2016-07-20:
Do people use this much? (Just as an estimate for priorities.) My intuition would be like @dengemann 's

agramfort, 2016-07-20:
low priority I agree but maybe we should document it better to attract more
users of this...

jona-sassenhagen, 2016-07-20:
Couldn't it be reconstructed by relying on e.g. sklearn's LOO cv? Building on standard tools rather than a custom implementation with a novel API?

agramfort, 2016-07-20:
yes probably...

closed 2016-07-26 by agramfort