mne-tools/mne-python #3432: Unhelpful error message for mne.coreg.gui when files are missing

jona-sassenhagen, 2016-07-18:
I tried to plot a subject with `mne.coreg.gui` for which I had not downloaded all surface files yet. I got this error message:

```
---------------------------------------------------------------------------
TraitError                                Traceback (most recent call last)
<ipython-input-3-b9d2c35f6324> in <module>()
      1 mne.gui.coregistration(
      2     subject='HA0407',
----> 3     subjects_dir="/Volumes/INTENSO/subjects/",
      4              )

/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/gui/__init__.pyc in coregistration(tabbed, split, scene_width, inst, subject, subjects_dir)
     65     from ._coreg_gui import CoregFrame, _make_view
     66     view = _make_view(tabbed, split, scene_width)
---> 67     gui = CoregFrame(inst, subject, subjects_dir)
     68     gui.configure_traits(view=view)
     69     return gui

/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/gui/_coreg_gui.pyc in __init__(self, raw, subject, subjects_dir)
   1265
   1266         if subject is not None:
-> 1267             self.model.mri.subject = subject
   1268
   1269         if raw is not None:

/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/traits/trait_types.pyc in _set(self, object, name, value)
   1964             self.set_value( object, name, value )
   1965         else:
-> 1966             self.error( object, name, value )
   1967
   1968 class Enum ( BaseEnum ):

/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/traits/trait_handlers.pyc in error(self, object, name, value)
    167         """
    168         raise TraitError( object, name, self.full_info( object, name, value ),
--> 169                           value )
    170
    171     def arg_error ( self, method, arg_num, object, name, value ):

TraitError: The 'subject' trait of a MRISubjectSource instance must be 'fsaverage', but a value of 'HA0407' <type 'str'> was specified.
```

Took me a while to figure out what the problem was.
Can't really reproduce because I don't know what files hadn't finished downloading yet ... but basically, mne.coreg.gui should give a more helpful error message when files are missing.

(Not the issue I mentioned on gitter Eric)

christianbrodbeck, 2016-07-19:
This sounds like we should update `_is_mri_subject()` to check for all the files that are actually needed, so in your case subjects_dir would have looked empty?

Eric89GXL, 2016-07-19:
> This sounds like we should update _is_mri_subject() to check for all the files that are actually needed, so in your case subjects_dir would have looked empty?

That sounds reasonable to me. Is there some way we can take it a step further to tell if *probably is meant to be a subject* but *definitely isn't usable*, and warn for that set of folders? For example if the standard folders are present like `mri`, `surf`, and `bem` but no head surface is found?

christianbrodbeck, 2016-07-19:
Yeah we could look for "partial" subjects and show an info dialog if there are any?

Eric89GXL, 2016-07-19:
Sounds good to me, although if it's a modal dialog then people with odd setups will get annoyed at having to click it away every use. Are you worried that a `warn(...)` will be missed?

christianbrodbeck, 2016-07-19:
Yeah that's true... we could just `warn()`, assuming that when something goes not as expected people will look to the Terminal

jona-sassenhagen, 2016-07-20:
Ideally, there'd be a message of what files specifically are missing, and what should be done to fix them (ie., re-run recon-all or whatever).

Not sure if this is practically, I don't really understand the MRI backbone so far.

christianbrodbeck, 2016-10-13:
https://github.com/mne-tools/mne-python/pull/3644 updated `_is_mri_subject()` to only look for the head files that are actually needed.

Eric89GXL, 2016-10-13:
I'll go ahead and close then, @jona-sassenhagen feel free to reopen if it's not sufficient

closed 2016-10-13 by Eric89GXL