mne-tools/mne-python #3433: BUG: examples/decoding/plot_decoding_xdawn_transformer.py fails on `master`

Eric89GXL, 2016-07-18:
cc @kaichogami @kingjr looks like a PR was merged with a failing example. Can you have a look?

Eric89GXL, 2016-07-18:
This is the traceback, CircleCI has the same error:
```
examples\decoding\plot_decoding_xdawn_transformer.py:32: RuntimeWarning: The default output type is "ba" in 0.13 but wil
l change to "sos" in 0.14
  raw.filter(1, 20, method='iir')
The default output type is "ba" in 0.13 but will change to "sos" in 0.14
Traceback (most recent call last):
  File "examples\decoding\plot_decoding_xdawn_transformer.py", line 47, in <module>
    score = cross_val_score(clf, X, y, cv=5)
  File "D:\Anaconda2\lib\site-packages\sklearn\cross_validation.py", line 1352, in cross_val_score
    cv = _check_cv(cv, X, y, classifier=is_classifier(estimator))
  File "D:\Anaconda2\lib\site-packages\sklearn\cross_validation.py", line 1606, in _check_cv
    cv = StratifiedKFold(y, cv, indices=needs_indices)
  File "D:\Anaconda2\lib\site-packages\sklearn\cross_validation.py", line 434, in __init__
    label_test_folds = test_folds[y == label]
IndexError: too many indices for array
```

kingjr, 2016-07-18:
Ok

kaichogami, 2016-07-19:
@Eric89GXL this is because `label_binarize` coverts `y` into a column
vector.
Even after correcting that, it will likely fail as `LogisticRegression`
expects a two dimensional matrix. I have addressed the issue in pr #3409.

kingjr, 2016-07-19:
I have updated the example in https://github.com/mne-tools/mne-python/pull/3425.

closed 2016-07-21 by Eric89GXL