mne-tools/mne-python #3435: TVTK requirement for building high res surfs

jona-sassenhagen, 2016-07-18:
Running `mne make_scalp_surfaces` on a Mac, I get

```
3. Creating medium tessellation...
3.1 Decimating the dense tessellation...
Traceback (most recent call last):
  File "/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/commands/mne_make_scalp_surfaces.py", line 144, in <module>
    run()
  File "/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/commands/mne_make_scalp_surfaces.py", line 56, in run
    options.verbose)
  File "<string>", line 2, in _run
  File "/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/utils.py", line 667, in verbose
    return function(*args, **kwargs)
  File "/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/commands/mne_make_scalp_surfaces.py", line 127, in _run
    n_triangles=n_tri)
  File "/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/surface.py", line 834, in decimate_surface
    return _decimate_surface(points, triangles, reduction)
  File "/Users/jona/anaconda3/envs/py2/lib/python2.7/site-packages/mne-0.13.dev0-py2.7.egg/mne/surface.py", line 791, in _decimate_surface
    raise ValueError('This function requires the TVTK package to be '
ValueError: This function requires the TVTK package to be installed
```

Eric89GXL, 2016-07-18:
This should only happen if decimation is requested, I'll fix it

Eric89GXL, 2016-07-18:
...looks like decimation is mandatory, I'll probably make an argument to supply the decimation levels

Eric89GXL, 2016-07-18:
Or since they're called "medium" and "sparse", maybe just `--nodecimate` would do it

PR #3438 (2016-07-20): MRG: Dont require TVTK
Closes #3435.

Adds an option not to decimate. @dengemann what do you use the other decimations (medium, sparse) for? I don't think they're part of the standard processing scheme...?
closed 2016-07-20 by Eric89GXL