mne-tools/mne-python #3441: Stickers?

choldgraf, 2016-07-19:
For a totally non-coding-related question:

I just got back from Scipy and there were so many cool stickers there for the various open-source projects in the scientific python ecosystem. Now, I'm not usually a "fill my laptop with stickers" kinda guy, but the scipy folks guilt tripped me into putting on a few of them.

Now that MNE has a fancy shmancy logo with nice colors and such, did anybody happen to make stickers for it? Imagine, your computer could one day look like this!

[image]

agramfort, 2016-07-19:
:)

kingjr, 2016-07-20:
lol #PhotoshopSkills

teonbrooks, 2016-07-20:
would be cool if our logo had an optional, transparent background. that would make for better printed decals.

Eric89GXL, 2016-07-20:
I'm pretty sure @drammock included scripts with the logo when he made it so you should be able to recreate it, but it's already an SVG somewhere so changing the background color in something like Inkscape should hopefully be trivial

teonbrooks, 2016-07-20:
cf. https://github.com/mne-tools/mne-python/pull/2395

I tried it a while back but I'm missing a font in mpl and that was a rabbit hole I couldn't go down. nevertheless, it still rendered

choldgraf, 2016-07-20:
Printing a clear MNE logo was pretty easy for me. Just modify the script to save to .svg and delete the background in Illustrator/whatever (or you might be able to just sit the background patch `visible` parameter to false in matplotlib). E.g.:

[image]

FWIW, all the other folks seemed to be using stickermule for their stickers. Would probably just be a matter of pooling together some cash to pay for 'em.

jona-sassenhagen, 2016-07-20:
That's actually the first time Donald Trump has "worked" for me.

If his job was running around and selling people on MNE, ... I think he'd be well suited to that.

Eric89GXL, 2016-07-20:
There is something wrong with that one... I think it's the wrong font perhaps

jona-sassenhagen, 2016-07-20:
The way the gradient works, it makes Trump's cheeks look really hamsterish.

agramfort, 2016-07-20:
just make a proper one and I'll find the cash :)

choldgraf, 2016-07-20:
i actually do think my computer is missing a font as well (probably the same problem @teonbrooks ran into). I think it's pretty easy to add the font to MPL if I can find the TTL file somewhere.

BTW, where is that logo-generating script these days? I had to find it in a link that somebody sent in an old PR, but I couldn't find it in the main codebase

choldgraf, 2016-07-21:
Does this look correct?

[image]

Eric89GXL, 2016-07-21:
That would looks a bit stretched, it should look like it does on the MNE-Python website:
[image]

choldgraf, 2016-07-21:
Hmm, well I'm using the script that was linked here:

https://github.com/mne-tools/mne-python/blob/d42a711f5184f64a251398bc39834cf46ac60d17/doc/generate_mne_logos.py

Only changing the output to `svg` instead of `png`. There there system /
software requirements for that script to work properly? Or is there an
updated version somewhere? It looks like it's not in master at that
location.

Eric89GXL, 2016-07-21:
I just generated this (on Windows even!) after:

1. installing [Primetime TTF] (http://www.dafont.com/primetime.font) and [Cooper Hewitt OTF] (https://www.cooperhewitt.org/wp-content/uploads/fonts/CooperHewitt-OTF-public.zip)
2. Removing `~/.matplotlib/fontList.cache` to force it to rebuild my cache
3. Removing `op.join`s in the writing (since I was running it from out of place)

and got the correct logo in PNG:

[image]

When modifying to save as SVG, I get the same output when I view it in Chrome (but can't attach here because GitHub doesn't support it).

Maybe you are getting matplotilb warnings about not finding the correct fonts when you run it?

I have no idea why that script is missing from `master`, I see it in the commit log, then it disappears at the next `master` commit for some reason.

Eric89GXL, 2016-07-21:
Oh actually the script is here (hah!):

https://github.com/mne-tools/mne-python/blob/master/logo/generate_mne_logos.py

Eric89GXL, 2016-07-21:
(found using http://stackoverflow.com/questions/6839398/find-when-a-file-was-deleted-in-git)

drammock, 2016-07-21:
Sorry to chime in so late.  @Eric89GXL is right, the script needs a couple of uncommon (though free) fonts to work properly.  Here's an SVG version:

http://dan.mccloy.info/data/mne-logo.svg
-- dan

choldgraf, 2016-07-21:
Huh, I wonder why it's appearing stretched on my computer. I'm not getting any font errors but it still looks slightly vertically stretched (using the version that is in the `logo` folder, which I stupidly missed before)

choldgraf, 2016-07-21:
Though either way it sounds like the SVG @drammock linked is all we need. Here's a PNG version w/o a background:

[image]

It looked like this is what the other folks at scipy were using:

https://www.stickermule.com/products/die-cut-stickers

drammock, 2016-07-22:
@all please note that the Primetime font used for the letters MNE is "free
for personal use only" so the stickers should not be used for fundraising /
profit.  The tagline font (Cooper Hewitt) is under a SIL Open Font Licence.

@choldgraf I've often noticed that matplotlib is inconsistent on when /
whether its missing font warnings make it through to stdout, so I wouldn't
be surprised if on your system it's failing to find Primetime and falling
back silently to DejaVu Sans.

choldgraf, 2016-07-22:
Yeah I'm not sure what's up with matplotlib. I was getting font errors before, but I found/downloaded the ttl files for those fonts and now I'm not getting the errors. Comparing the PNG I showed above vs. the correct ones below it, the fonts look pretty much the same, so I'm not sure what's going on

Eric89GXL, 2016-09-08:
@agramfort WDYT about making a handful of these to keep with our poster at BioMag?

choldgraf, 2016-09-09:
I can bring some to SfN as well if they're made before then. If you want to ship some to NYC then I can pick them up there (I'll be there in October)

agramfort, 2016-09-09:
@yousrabk @jasmainak can you maybe take care of this? I will pay and manage to get reimbursed afterwards

jasmainak, 2016-09-09:
@agramfort how many and what size? is 2" x 2" the size that is typical size?

choldgraf, 2016-09-09:
@jasmainak for reference, the two stickers in the dopey-looking picture of me at the top of this thread are 3" x 1" and 3" x 1.5", respectively. I think it's good to have at least some in that size since it's the common laptop sticker size.

agramfort, 2016-09-10:
I would make about 150. wdyt?

fine with 3" x 1.5"

jasmainak, 2016-09-10:
okay we might have some stickers by 23rd sept ;)

dengemann, 2016-09-10:
LOL .... If someone would have told me 4 years ago that one day we'd be
discussing stickers on github. Good stuff!

jasmainak, 2016-09-10:
are we happy with this?
[image]
It's going to be 3" x 1.3542". Speak up now if you have any reservations! Otherwise I'll approve it tomorrow morning.

dengemann, 2016-09-10:
It reminds me of Nintendo game cartridges from the early 90's .... Awesome!

Eric89GXL, 2016-09-10:
Awesome

jasmainak, 2016-09-10:
okay since I got two +1s, I'm going to approve it now :)

choldgraf, 2016-09-10:
Do you think you could get it at 1.3543" instead?

jasmainak, 2016-09-10:
I rather like the trailing 42 ;)

choldgraf, 2016-09-10:
haha, shame on me for not seeing the answer to the universe when it's staring me in the face

kingjr, 2016-09-12:
nice

jasmainak, 2016-09-20:
closed by https://twitter.com/mainakjas/status/778153289150062593

closed 2016-09-20 by jasmainak

Eric89GXL, 2016-09-20:
Nice, closing via tweet. Next issue we should closing via vine

teonbrooks, 2017-02-07:
So I was just informed that our stickers do not conform to cool kids standards. Apparently [hexagonal stickers](www.hexb.in) are in. Should we do stickers 2.0?

choldgraf, 2017-02-07:
omg hexagons are so 2016, I hear octagons are the new way to go. So I propose being even more cutting edge and going with a dodecahedron.

kingjr, 2017-02-07:
> standards. Apparently hexagonal stickers <http://www.hexb.in> are in.

ha!

Eric89GXL, 2017-02-08:
Shall we reformulate as a hexagon? We could do it in time for the March sprint

choldgraf, 2017-02-08:
fine w/ me, shouldn't be too expensive or time consuming. Maybe I can get BIDS to pay for some.

also this one is great

[image]

dengemann, 2017-02-08:
icosahedron!

jasmainak, 2017-02-08:
if you order again through stickermule, let me know. I might have a discount code lying around somewhere. Not sure though ...

teonbrooks, 2017-02-09:
hexagon! hexagon! @bmpvieira, you started something ;)

drammock, 2017-02-11:
Here you go, everyone: https://gist.github.com/drammock/2020b5111135bafb71ce6b5d8488495f

Eric89GXL, 2017-02-13:
Hah, great!

choldgraf, 2017-02-13:
I like it! I am a little concerned that the text underneath will be hard to read. Have you played around with making the font a bit stronger? Maybe it's fine when printed tho...

drammock, 2017-02-13:
@choldgraf no I didn't...  I was imagining a larger size, but I just checked back and discovered the [sticker constructor spec](https://github.com/terinjokes/StickerConstructorSpec).  Will revise and update with those dimensions in mind.

choldgraf, 2017-02-13:
cool - I think that is quite close though :) we should also make sure that
having the lettering so near the edges of the hexagon won't mean that it
gets cut off or overwritten by a line or something

drammock, 2017-02-14:
@choldgraf [the gist](https://gist.github.com/drammock/2020b5111135bafb71ce6b5d8488495f) has been updated to

1. conform to constructor spec
2. pull wordmark in from edge of hexagon a bit
3. boldify tagline

Send some to Seattle when they're ready!

choldgraf, 2017-02-14:
has anyone here printed hexagonal stickers before? I looked at the main company that website suggests and it isn't clear since I don't read japanese (also...the website header is very japanese and it's kind of great)

[image]

Eric89GXL, 2017-02-15:
Nope, no idea about where to do it

choldgraf, 2017-03-07:
just FYI I ordered about 60 of these, will bring 'em to NYC

teonbrooks, 2017-03-10:
@choldgraf how did it turn out, layout-wise?

choldgraf, 2017-03-10:
it looked like a hexagon, in the end ;-)

choldgraf, 2017-03-10:
jk, here's the proof: https://www.stickermule.com/artworks/954501?token=7aba51b4d5450dc7314b320219c276b6

[2 more replies]