mne-tools/mne-python #3442: ENH: GSoC 2016

kingjr, 2016-07-20:
We have quite a lot ongoing PRs, so I thought I would try to organize them here to keep track of the big picture. I'll edit the post along the issue along the way.

The aim is to make transformers that follow the sklearn API:

```
pipe = make_pipeline(
    Filterer(sfreq=200, None, 30),
    Xdawn(),
    SearchLight(make_pipeline(StandardScaler(), LogisticRegression())
)

score = cross_val_score(X=epochs.get_data(), y=epochs.event[:, 2])
```

For now, we're focusing on sklearn integration, not high level features (plotting, get_coefs_ etc).

* Spatial filters:
    - [ ] XdawnTransformer: started in #3245, refactoring in #3425
    - [X] CSP: only needs a renaming of `epochs_data` into `X`.
    - [ ] SSP
    - [ ] Generic spatial filter: started in #3276

* Freq - time/freq transformers:
    - [ ] FilterEstimator: needs refactor from `mne.decoding.FilterEstimator` to pass explicit args and not `info` + BUG: #3395
    - [X] PSDEstimator
    - [ ] TimeFrequencyDecomposer: see (here)[https://github.com/kingjr/jr-tools/blob/master/jr/gat/transformers.py]

* Single trial covariances
    - [ ] Covariances (current `mne.cov.Cov` not compatible with sklearn)
    - [ ] Other from pyRiemann? `TangentSpace`, `HankelCovariances`.

* Continuous signals (raw not epochs)
    - [ ] Xdawn overlap: to be integrated with the RERP code #2332
    - [ ] STRF: started in #3310

* Search lights
    - [ ] Generic SearchLight and GeneralizationLight: to be reviewed in #3381
    - [ ] Refactor TimeDecoding & GeneralizationAcrossTime
    - [ ] ENH: Time Frequency Decoding
    - [ ] EMS: #3427

* Preprocessing
    - [ ] Scaler: scale channels over all time their corresponding time points to deal with issues related to mix channel types needs refactoring to comply to `X` `y` API
    - [ ] BadChannelRemover: remove bad channels to avoid linear alg errors: e.g. https://github.com/alexandrebarachant/pyRiemann/pull/30
    - [ ] Vectorizer: to pass from n-D `X` to 2D `X` to be reviewed in https://github.com/mne-tools/mne-python/pull/3409
    - [ ] TimePadder: to minimize edge artefact: e.g. start [here](https://github.com/kingjr/jr-tools/blob/master/jr/gat/transformers.py)
    - [ ] Reshaper: to apply search lights in ad hoc combination of dimension (e;g. time frequency) e.g. start [here](https://github.com/kingjr/jr-tools/blob/master/jr/gat/transformers.py)

* Plot
    - [ ] Mixin class to plot patterns_ and filters_? see https://github.com/mne-tools/mne-python/issues/3424

* Docs
    - [ ] Tutorials
    - [ ] Example
    - [ ] Organization of the modules: #3428

cc @kaichogami @agramfort @dengemann

Eric89GXL, 2017-03-21:
@kingjr is this issue still the roadmap, or should we close?

kingjr, 2017-03-21:
Still rough roadmap, although needs to be updated. Let's close it after the sprint IMO.

kingjr, 2017-03-31:
I'm actually for keeping it up ;)

jona-sassenhagen, 2017-04-03:
@kingjr , @agramfort 's idea is one which will result in rERP and the receptive field module staying separate forever.

- rERP becomes restricted to the original idea, allowing only sparse predictors; dense predictors are handled by receptive field module
- receptive field module should be extended slightly to simply also using sparse predictors
(- XDawn and rERP should still be united)

jona-sassenhagen, 2018-08-04:
I closed/crossed out a few that had been done since.
8 issues still open.

jona-sassenhagen, 2018-08-04:
@kingjr any idea for an example for UnsupervisedSpatialFilter with the example data? The only thing I can think of is show that you can use an unregularized algorithm when p > n by reducing p in dimensionality via PCA, i.e., for the MEG data with many channels and few trials.

kingjr, 2018-08-04:
How about PCA and then sliding with svm multiclass on the left/right
audio/visual dataset ?