mne-tools/mne-python #3444: mne.event.define_target_events; AttributeError

hoirkmann, 2016-07-20:
Hello there,

when I run mne.event.define_target_events the following traceback is displayed:
`Traceback (most recent call last):
  File "compute_1.5.py", line 187, in <module>
    events = mne.event.define_target_events(events=events,reference_id=cond,target_id=99,sfreq=500.0,tmin=0.0,tmax=2.0,new_id=cond,fill_na=None)
  File "/home/jon/.local/lib/python3.4/site-packages/mne/event.py", line 122, in define_target_events
    for event in events.copy().astype(int):
AttributeError: 'tuple' object has no attribute 'copy'`

I'm pretty sure, the args are in order. events is the required object as returned by mne.find_events. cond is an integer matching the markers in the data. new_id is set to cond to include events containing the target-marker and keep their code, drop those not containing the target.

My mne-version is 0.11.0 .

Thanks for your help and this great package in general,
Jonas

jona-sassenhagen, 2016-07-20:
What does the `events` object look like when you print it?

Eric89GXL, 2016-07-20:
`define_target_events` returns two elements (i.e., a two-element tuple):

http://mne-tools.github.io/stable/generated/mne.event.define_target_events.html#mne.event.define_target_events

If you do `events = define_target_events(...)[0]` to pick the first one or `events, lag = define_target_events(...)` you should be good to go

closed 2016-07-20 by Eric89GXL