mne-tools/mne-python #3451: ica.plot_overlay - red and black lines are now mixed

mmagnuski, 2016-07-23:
[`plot_overlay` mixes red and black lines](http://www.martinos.org/mne/dev/_images/sphx_glr_plot_artifacts_correction_ica_010.png) because `mne.viz.evoked._plot_evoked` uses `z_order` param now and the default 'unsorted' value gives different z-order to each line. When two evokeds are plotted successively (before ICA - red; after ICA - black) this causes the lines to mix.
Potential solutions:
* `_plot_evoked` could expose another z_order option ('no', 'none', same', 'flat'?) that does not set line z-order
* `_plot_evoked` 'unsorted' z_order could lead to not changing z-order at all
* `ica.plot_overlay` could use the option to pass a callable to z_order where the callable would just return `np.ones(n_channels)`

I think the first and second solutions are more general and thus better.

jona-sassenhagen, 2016-07-23:
In https://github.com/mne-tools/mne-python/blob/master/mne/viz/evoked.py#L348

`zorder=z+1 if spatial_colors else 1`

?

mmagnuski, 2016-07-23:
@jona-sassenhagen even better :)

agramfort, 2016-07-23:
PR welcome

PR #3453 (2016-07-25): MRG fix for #3451
solution suggested by @jona-sassenhagen
closes #3451
commit 6957b937 (2016-07-25): MRG fix for #3451 (#3453)

* fix for #3451

solution suggested by @jona-sassenhagen
closes #3451

* line was too long
closed 2016-07-25 by agramfort