mne-tools/mne-python #3459: mne tries to download data to nonexistent directory (it tries Anaconda not Miniconda3 directory)

mmagnuski, 2016-07-26:
when doing:
```python
from mne.datasets import sample
data_path = sample.data_path()
```
I get the following error:
```
Downloading or reinstalling data archive MNE-sample-data-processed.tar.gz at location C:\Anaconda\lib\site-packages\examples
Downloading data from https://mne-tools.s3.amazonaws.com/datasets/MNE-sample-data-processed.tar.gz (1.35 GB)

Error while fetching file https://mne-tools.s3.amazonaws.com/datasets/MNE-sample-data-processed.tar.gz. Dataset fetching aborted.

FileNotFoundErrorTraceback (most recent call last)
<ipython-input-2-9e480f3f6cd3> in <module>()
      8 
      9 # getting some data ready
---> 10 data_path = sample.data_path()
     11 raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
     12 

e:\proj\src\mne-python\mne\datasets\sample\sample.py in data_path(path, force_update, update_path, download, verbose)

e:\proj\src\mne-python\mne\utils.py in verbose(function, *args, **kwargs)
    668         with use_log_level(verbose_level):
    669             return function(*args, **kwargs)
--> 670     return function(*args, **kwargs)
    671 
    672 

e:\proj\src\mne-python\mne\datasets\sample\sample.py in data_path(path, force_update, update_path, download, verbose)
     20     return _data_path(path=path, force_update=force_update,
     21                       update_path=update_path, name='sample',
---> 22                       download=download)
     23 
     24 data_path.__doc__ = _data_path_doc.format(name='sample',

e:\proj\src\mne-python\mne\datasets\utils.py in _data_path(path, force_update, update_path, download, name, check_version, return_version, archive_name)
    264             if fetch_archive:
    265                 _fetch_file(url, archive_name, print_destination=False,
--> 266                             hash_=hash_)
    267 
    268         if op.exists(folder_path):

e:\proj\src\mne-python\mne\utils.py in _fetch_file(url, file_name, print_destination, resume, hash_, timeout, verbose)

e:\proj\src\mne-python\mne\utils.py in verbose(function, *args, **kwargs)
    668         with use_log_level(verbose_level):
    669             return function(*args, **kwargs)
--> 670     return function(*args, **kwargs)
    671 
    672 

e:\proj\src\mne-python\mne\utils.py in _fetch_file(url, file_name, print_destination, resume, hash_, timeout, verbose)
   1691         scheme = urllib.parse.urlparse(url).scheme
   1692         fun = _get_http if scheme in ('http', 'https') else _get_ftp
-> 1693         fun(url, temp_file_name, initial_size, file_size, verbose_bool)
   1694 
   1695         # check md5sum

e:\proj\src\mne-python\mne\utils.py in _get_http(url, temp_file_name, initial_size, file_size, verbose_bool)
   1600                            verbose_bool=verbose_bool)
   1601     chunk_size = 8192  # 2 ** 13
-> 1602     with open(temp_file_name, mode) as local_file:
   1603         while True:
   1604             t0 = time.time()

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Anaconda\\lib\\site-packages\\examples\\MNE-sample-data-processed.tar.gz.part'
```

This happens on dev mne-python (almost latest master, about 3 commits behind).
The following command:
```python
mne.utils.get_config()
```
returns:
```
{'MNE_BROWSE_RAW_SIZE': '24.0,11.45',
 'MNE_DATA': 'C:\\Anaconda\\lib\\site-packages\\examples'}
```
but because I have miniconda installed on this computer, the correct path should begin with `C:\Miniconda3\Lib\site-packages`.

Other than that, I don't know what specifically goes wrong, but will happily try out any ideas.

agramfort, 2016-07-26:
current master should download stuff in your home dir now rather than in
the install path.

mmagnuski, 2016-07-26:
updated to current master - still the same error.

agramfort, 2016-07-26:
this PR https://github.com/mne-tools/mne-python/pull/3437
should have fixed it :(

Eric89GXL, 2016-07-26:
You probably have an old entry in your config file, can you try
mne.get_config() and report the result?

I'm not sure how master could otherwise point to the examples directory but
maybe I missed it. Can you try a little bit of debugging?

mmagnuski, 2016-07-26:
I will try some debugging, the output of `get_config` is in my original post, but I'm pasting it here:
```python
mne.utils.get_config()
```
returns:
```
{'MNE_BROWSE_RAW_SIZE': '24.0,11.45',
 'MNE_DATA': 'C:\\Anaconda\\lib\\site-packages\\examples'}
```
so I seem to have an old entry in config. It makes sense - I might have had Anaconda before Miniconda, but mne config stayed where it was. So maybe adding  some check for existence of the directory and some meaningful error message would be good?

Eric89GXL, 2016-07-27:
I thought it tried to do some writing, but maybe it doesn't. I guess we could try writing a temp file to the directory and see if it passes.

mmagnuski, 2016-08-13:
I deleted the old config and everything worked fine. This should be a very rare problem but test-writing to the dir and some helpful error message (asking user to check its config, dir existence etc.) when this fails would be cool.

agramfort, 2016-08-14:
+1 for a PR on docstrings improvements

closed 2016-08-19 by Eric89GXL