mne-tools/mne-python #3460: BUG: ICA mixing matrix shape mismatch

agostontorok, 2016-07-26:
I recently got into a small trouble:

Description: When I set _n_ for the value of n_components in and ICA decomposition and _n_ < Picks, it results in an ica object with a mixing_matrix_.shape of _n_ by _n_ instead of _n_ by Picks.

Here is a toy code to reproduce the bug:

```python
import mne
from mne.preprocessing import ICA
from mne.datasets.brainstorm import bst_raw
mne.set_log_level('INFO')

print(__doc__)

tmin, tmax, event_id = -0.1, 0.3, 2  # take right-hand somato
reject = dict(eog=250e-6)

data_path = bst_raw.data_path()

raw_fname = data_path + '/MEG/bst_raw/' + \
                        'subj001_somatosensory_20111109_01_AUX-f_raw.fif'
raw = mne.io.read_raw_fif(raw_fname, preload=True, add_eeg_ref=False)

ica = ICA(n_components = 10, method = "fastica") # 
S_ = ica.fit(raw, start = 0, stop = 100, verbose = True)
S_.mixing_matrix_.shape # this is (10, 10) instead of (100, 10)
```

BTW mne.__version__ is '0.12.0'

Best,

Agoston

agramfort, 2016-07-26:
this is not a bug; The mixing matrix is used after dim reduction via PCA, so it's always squared. You need to take into account the pca_components_ attribute. See the code if needed.

reopen if needed

closed 2016-07-26 by agramfort

agostontorok, 2016-07-26:
The `S_.pca_components_.shape` is (100, 274) in this example. Note I did not set either n_pca_components or max_pca_components. The description says:
```python
"""max_pca_components : int | None
        The number of components used for PCA decomposition. If None, no
        dimension reduction will be applied and max_pca_components will equal
        the number of channels supplied on decomposing data. Defaults to None."""```

Based on this I would think that no PCA was done before. In any case, I do not find an attribute that informs me about the weights of individual channels on an ICA component, or at least converts between the PCA result matrix and the ICA mixing matrix (100 x 274 vs. 10 x 10).

agramfort, 2016-07-26:
see for example here:

https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L1270

after picking the good ica sources we put back the PCA components up to a
number given by:

_n_pca_comp = self._check_n_pca_components(self.n_pca_components)

agostontorok, 2016-07-26:
Thank you Alexandre for the reference, I see, so the ICA is done on the _n_components_ PCs with the highest explained variance. But can I run the analysis __without__ PCA? I tried setting both n_pca_components = None, max_pca_components = None and still the result was w/ PCA

mmagnuski, 2016-07-26:
If `n_components` is None all PCA components should be used in ICA, so it is like without PCA dim reduction.

jona-sassenhagen, 2016-07-26:
If you want to get the weight maps, use `mne.preprocessing.ica._get_ica_map`.

dengemann, 2016-07-26:
if `n_pca_components` is None, you use all PCA components, amounting to not
actually performing PCA but simply using it for sphering. `n_components`
refers to the ICA components.

mmagnuski, 2016-07-26:
@dengemann Right, I meant `n_pca_components`. But I think the interactions between `n_components`, `n_components_`, `n_pca_components` and `max_pca_components` may be a little confusing - what do you think about clarifying the docs a little?

jona-sassenhagen, 2016-07-26:
I think it might be worth to make get_ica_map a public method of ICA. We have had requests for the weight maps a few times now IIRC.

dengemann, 2016-07-26:
> what do you think about clarifying the docs a little

you're volunteering right? :)
Clearer docs are always a good idea.

mmagnuski, 2016-07-26:
@dengemann @jona-sassenhagen 
I'll try to take care of both in the next few days (I was diving into ICA code anyway to sort infomax components by explained variance).

dengemann, 2016-07-26:
>  (I was diving into ICA code anyway to sort infomax components by explained variance).

I guess that's a much desired feature by Infomax fans. That would be a nice contribution. @mmagnuski don't hesitate to blame me if you get stuck with ICA code hermeneutics.

agramfort, 2016-07-27:
looking at @jmontoyam 's code something like this:

```
    mvar = np.sum(linalg.inv(unmixing_) ** 2, axis=0) *\
       np.sum(Y ** 2, axis=1) / (N * T - 1)
    windex = np.argsort(mvar)[::-1]

    unmixing = unmixing_[windex, :]
```

seems to match the eeglab results.

HTH

agostontorok, 2016-07-27:
Thank you both for this discussion, I would be more than happy if you clarify the documentation. 

Just a few extra comments:
@dengemann, you wrote

> if `n_pca_components` is None, you use all PCA components, amounting to not
actually performing PCA but simply using it for sphering. `n_components`
refers to the ICA components.

based on this I would expect 274 PC of the 274 channels in the above example, yet I get 100. Also I checked when I set max_pca_components to 200 it is still 100 (??). however if I run [the underlying source code](https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L504) separately on the MEG data in the example using 200 as max_pca_components I get 200 PC (as expected). So @mmagnuski your clarification on the docs is appreciated in advance

mmagnuski, 2016-07-29:
@agramfort 
Thanks, that makes sense, just a quick question if I understand it correctly:
`Y` is the components signal (components x timesamples) and `T` and `N` are number of timesamples and number of channels?

I won't have reliable internet acess till monday, but I'll try to have something for a PR then.

agramfort, 2016-07-29:
yes correct. Although N and T have no impact on ordering.