mne-tools/mne-python #3464: ENH: make _segment_raw public

kingjr, 2016-07-27:
@mmagnuski suggested using mne.epochs._seg_raw, in the mne discussion, but the function is private. Shall we make it public?

mmagnuski, 2016-07-27:
There was a discussion about it during my plot_properties PR (where _segment_raw was added). I think it is pretty common usage and we even considered making a `segment` method for raw.
If I remember correctly @jona-sassenhagen was +1 but @agramfort and @Eric89GXL were not convinced so it ended up being private.
I'm +1 on making it public, but I don't want to start a fight ;)

Eric89GXL, 2016-07-27:
It's okay to start fights it's the only way to get us curmudgeonly folk to
budge sometimes :)

My recollection was that such a function would essentially call
make_fixed_length_events followed by Epochs constructor, thus end up with a
nasty signature if it wanted to provide the same amount of flexibility. The
only advantage would be it would use one function call instead of two,
right?

kingjr, 2016-07-27:
Then I would add brief section to the preprocessing tutorial to make these two calls

kingjr, 2016-07-27:
(by I, I actually don't mean I)

jona-sassenhagen, 2016-07-27:
IIRC didn't say it should be made public, but that it might be cool to have it as a method.

mmagnuski, 2016-07-27:
@Eric89GXL the function is definitelly short, removing the docstring it is just:
```python
@verbose
def _segment_raw(raw, segment_length=1., verbose=None, **kwargs):
    events = make_fixed_length_events(raw, 1, duration=segment_length)
    return Epochs(raw, events, event_id=[1], tmin=0., tmax=segment_length,
                  verbose=verbose, baseline=None, **kwargs)
```
but I still think it is easier to write and *remember* this instead:
```
segments = segment_raw(raw, 2.)
```

or along the lines of what @jona-sassenhagen says:
```
segments = raw.segment(2.)
```

I can make a small PR to fix this - whichever you prefer:
* adding a few lines to some example/tutorial to show how to use `make_fixed_length_events` and then pass the events to `mne.Epoch`
* make public `segment_raw`
* add `segment` method to raw (and maybe epochs too?)
* add `segment` function that can handle both Raw and Epochs (?)

agramfort, 2016-07-27:
can you explain me:

why you want to segment epochs?

what is your use case of segment? is it really that common that 2 function
calls is too much?

Eric89GXL, 2016-07-27:
What if we make it that if the events argument to Epochs a float-like, it
creates events of that duration with event number 1? Then it's only a
single call for people who need a shortcut.

mmagnuski, 2016-07-27:
Well, segmenting epochs isn't common, I recently had a few cases for that, but I agree it is not very useful.
Segmenting raw signal on the other hand is pretty common, I think, if you want to perform manual or automatic rejection on continuous data. 
Another, related use-case: back in matlab-only times I would segment raw data, reject segments too far from the task of interest, reject strong artifacts and feed the remaining segments to ICA - this way I usually had much more data for ICA (especially if trials had big differences in length - because for example cue-target interval was of experimental interest).

But I do well with using `_segment_raw` so I can just do option 1. :)

mmagnuski, 2016-07-27:
@Eric89GXL It would be ok, I think, but would have to override tmin, tmax and possibly baseline (otherwise for events=2. and therefore tmin=0 and tmax=2, the default baseline would be done with respect to one sample (t==0))

Eric89GXL, 2016-07-27:
>  but would have to override tmin, tmax and possibly baseline (otherwise for events=2. and therefore tmin=0 and tmax=2, the default baseline would be done with respect to one sample (t==0))

We could easily set `tmax=None`, which would be 1. for traditonal events and whatever float `events` for this scheme. The baseline indeed wouldn't make much sense, but really there shouldn't be a baseline anyway here since the data are not time locked to anything. In other words we could have `baseline='auto'` mean `(None, 0)` for traditional events and `None` for float `events`. This does start to feel a bit too magical, though.

agramfort, 2016-07-28:
I like @EricGLX89 suggestions with events=2. but keep the magic low... I
feel that setting 3 parameters is ok...

dengemann, 2016-07-28:
I don't understand this entire debate.
We have Epochs and on the other hand ICA will do an online segmentation for
you.
Is there anything that cannot be done by creatively using Epochs?

mmagnuski, 2016-08-12:
Sorry I didn't respond. I didn't mean it's impossible, but that using Epochs for segmentation is currently not very convenient. I am using `_segment_raw` so I don't have any problem with that, I just thought it might be useful to have it public. 
Anyway - this can be closed. The issue is not worth the debate it provoked.

closed 2016-08-12 by kingjr