mne-tools/mne-python #3465: Transform docs and API

jona-sassenhagen, 2016-07-27:
There should be

- good documentation
- accessible functions

for transforming between the various coordinate systems, e.g. volume to source, sensor to RAS, etc.

@Eric89GXL @christianbrodbeck

Eric89GXL, 2016-07-27:
And also the relationship between high-density cortical meshes (i.e., what FreeSurfer gives us) across different subjects and source spaces (decimated versions of those).

Eric89GXL, 2016-07-27:
Currently the best "documentation" of these is in the `setup_volume_source_space` function because most of these transforms needed to be used there. There is also a little bit in the `SourceEstimate` class IIRC having to do with getting to Talairach.

christianbrodbeck, 2016-07-27:
Related to this, a few weeks ago I wrote a function to coregister head to MRI programmatically based only on the three fiducials, I don't know if it's desirable to include but if it is I can make a PR

jona-sassenhagen, 2016-07-27:
Jesus H. Christ yes. That's precisely what I need. How soon can you PR? :)

christianbrodbeck, 2016-07-27:
It's in #3466

Eric89GXL, 2017-02-14:
I added these doc things to #1495, closing

closed 2017-02-14 by Eric89GXL