mne-tools/mne-python #3467: ENH: manual edition of annotations & channels

deep-introspection, 2016-07-28:
Hi there,

it could be nice to manually (add/delete/modify) annotations and (status/type) of channels.
On top of being convenient, it would also help many people transitioning to MNE Python.

The MNE annotations seem to quite suit the job for the former part, but such feature would maybe require to expand the GUI of the interactive plot or build something dedicated.

Did you already thought about something like that? Are people working on it?

Cheers!
Guillaume

agramfort, 2016-07-28:
yes allowing to create and modify in the raw data browser is planned

deep-introspection, 2016-07-28:
Cool! How can I help?

agramfort, 2016-07-28:
hack the raw matplotlib plotting code :)

agramfort, 2016-07-28:
and submit a clean PR :)

closed 2017-01-30 by agramfort