mne-tools/mne-python #3471: ENH: Filterer for decoding

kingjr, 2016-07-29:
@kaichogami 

We need to create a `mne.decoding.Filterer` object that takes `low'`, `high`, `sfreq` ... params, deals with 3D numpy array (n_epochs, n_chans, n_times) and transform the data with low/high/band pass filters.

It would be great to add a time_padding option to add zeros before and after tmin and tmax, so as to minimize edge artefact.

We'll eventually deprecate `FilterEstimator` in one or two deprecation cycles, as it uses `info` as param.

Eric89GXL, 2016-07-29:
Keep in mind the IIR and FIR both already do (different forms of) padding

kingjr, 2016-07-29:
@Eric89GXL Ok thanks. How about the time frequency transform ?

Eric89GXL, 2016-07-29:
I can't remember, you'll have to look at the code

kingjr, 2016-08-02:
PR in #3472

kaichogami, 2016-09-10:
This can be closed now.

closed 2016-09-10 by agramfort