mne-tools/mne-python #3473: Cannot find channel types using "reject" (but "pick_types" works)

nicoa47, 2016-07-30:
I have an issue using reject based on eeg channels in mne.Epochs: It gives the error "No EEG channel found. Cannot reject based on EEG."

Here is an example file:
https://www.dropbox.com/s/i81x10ygemv5tbm/example_raw.fif?dl=0

Here is the relevant code:
`#load data
#this fif file was created from an EDF file, using MNE's save command
raw = mne.io.read_raw_fif('example_raw.fif', preload=True)

#set channel types (simply using "set_channel_types" gives an error)
raw.info['projs'] = []
raw.set_channel_types({'EMG l':'misc','EMG r':'misc','E1':'eog','E2':'eog','F3':'eeg','F4':'eeg','C3':'eeg','C4':'eeg','O1':'eeg','O2':'eeg'})
raw.add_proj(mne.preprocessing.ssp.make_eeg_average_ref_proj(raw.info, activate=False))

#check if channels set correctly
print('originally: \n', raw.info)
raw.pick_types(eeg=True)
print('after picking eeg: \n', raw.info)
#seems to work --> now only 6 channels left (but why called stim?)

#artifact rejection
chunk_size = raw.info['sfreq'] * 30
chunk_starts = np.arange(0, raw.n_times, chunk_size)
events = np.c_[chunk_starts, np.zeros((len(chunk_starts), 2))].astype(int)
epochs = mne.Epochs(raw, events[:-1], reject=dict(eeg=200e-06)) #error: no channels of type eeg
mne.viz.plot_epochs(epochs)
input()`

agramfort, 2016-07-30:
try this:

```
import numpy as np
import mne

raw = mne.io.read_raw_fif('example_raw.fif', preload=True)

#set channel types (simply using "set_channel_types" gives an error)
raw.info['projs'] = []
raw.set_channel_types({'EMG l':'misc','EMG r':'misc','E1':'eog','E2':'eog','F3':'eeg','F4':'eeg','C3':'eeg','C4':'eeg','O1':'eeg','O2':'eeg'})
raw.add_proj(mne.preprocessing.ssp.make_eeg_average_ref_proj(raw.info, activate=False))

#check if channels set correctly
print('originally: \n', raw.info)
raw.pick_types(eeg=True)
print('after picking eeg: \n', raw.info)
#seems to work --> now only 6 channels left (but why called stim?)

raw._data *= .1  # are data really in Volts?

#artifact rejection
chunk_size = raw.info['sfreq'] * 30
chunk_starts = np.arange(0, raw.n_times, chunk_size) + raw.first_samp
events = np.c_[chunk_starts, np.zeros((len(chunk_starts), 2))].astype(int)
epochs = mne.Epochs(raw, events[:-1], reject=dict(eeg=300e-06)) #error: no channels of type eeg
mne.viz.plot_epochs(epochs)
```

you had no good epoch. You did not take into account raw.first_samp and the reject was too low.

closed 2016-07-30 by agramfort

nicoa47, 2016-07-30:
Thank you, I did indeed miss that but it still gives me the same error at the second last line.

agramfort, 2016-07-31:
the code I pasted above works for me with current mne-python master version.

see http://martinos.org/mne/dev/install_mne_python.html#install-the-mne-python-package

if you need to work with development version of MNE