mne-tools/mne-python #3475: ENH: specify scoring in search light

kingjr, 2016-07-30:
Currently `SearchLight.score()` uses the scorer of the base_estimator.

Typically we would like to change the scoring, e.g. to use an AUC for a logistic regression, I was thinking of doing the following API:
```
sl = SearchLight(LogisticRegression(), scoring=my_scorer())
sl.fit(X, y).score(X, y)
```

An alternative is to change the estimator directly e.g.
```
clf = ChangerScorer(LogisticRegression(), my_scorer())
sl = SearchLight(clf)
sl.fit(X, y).score(X, y)
```

What I'm unclear is how to adapt the predict method as a function of the scoring metrics, and whether this should be done automatically. 

Anyone has a clear idea how what would be the most sklearn way of doing it?

agramfort, 2016-07-31:
have a look at this:

http://scikit-learn.org/stable/modules/generated/sklearn.metrics.make_scorer.html

regarding API I prefer option 1.

kingjr, 2016-07-31:
so in terms of API, we should ask users to use make_scorer, or shall it be done internally?

agramfort, 2016-07-31:
scoring can be a string if standard score. Else yes use make_scorer

kingjr, 2016-08-01:
@kaichogami would you like to give this a try?

kaichogami, 2016-08-01:
@kingjr Alright I will work on it.

kingjr, 2016-08-01:
@agramfort if we use cross_val_score, should scoring be defined at the highest level or within search light i.e.
1. `cross_val_score(make_pipeline(SearchLight(Ridge(), scoring=my_scorer()), X, y)`
or 
2. `cross_val_score(make_pipeline(SearchLight(Ridge()), X, y, scoring=my_scorer())`

agramfort, 2016-08-01:
2 is better

kingjr, 2016-08-01:
ok, so we need to handle both cases, right?: i.e.
*`SearchLight(Ridge(), scoring=my_scorer())`
*`cross_val_score(SearchLight(Ridge(), X, y, scoring=my_scorer())`

agramfort, 2016-08-02:
>
> ok, so we need to handle both cases, right?: i.e.
> *SearchLight(Ridge(), scoring=my_scorer())
>

what is the use case for this one? can't you rely on estimator.score?

>
> *cross_val_score(SearchLight(Ridge(), X, y, scoring=my_scorer())
>

+1 for this one.

kingjr, 2016-08-02:
> what is the use case for this one? can't you rely on estimator.score?

My use case is fit a search light on a localizer (no CV) and apply it to a main epoch dataset. However I typically want auc not accuracy. Maybe too specific ?

jona-sassenhagen, 2016-08-02:
>My use case is fit a search light on a localizer (no CV) and apply it to a main epoch dataset. However I typically want auc not accuracy. Maybe too specific ?

Confirming precisely this is a likely usage case.

agramfort, 2016-08-02:
ok point taken Use case is legit for me

kingjr, 2016-12-22:
closed by #3502 & #3833

closed 2016-12-22 by kingjr