mne-tools/mne-python #3482: Private API in Docs

christianbrodbeck, 2016-08-02:
Looking for docs on `Raw` indexing I noticed [this](http://martinos.org/mne/dev/auto_tutorials/plot_object_raw.html#indexing-data), isn't it bad practice to encourage use of a private attribute, especially since the proper API of `__getitem__` could be used as further below?

christianbrodbeck, 2016-08-02:
Also something that seemed slightly counterintuitive to me is that `mne.find_events()` returns `Raw` indexes with `raw.first_samp` added to them, however, `Raw.__getitem__` indexes data directly, ignoring `raw.first_samp`. As a consequence, using events for indexing `Raw` does not return the expected data, rather one would have to.

```py
event = mne.find_events(raw)[0]
data = raw[:, event[0] + raw.first_samp]
```

I could not see that in the docs, might be useful to add a note on it to the docs of `Raw.__getitem__`, `find_events` as well as the `Raw` Tutorial?

Eric89GXL, 2016-08-02:
It's better to document the existence somewhere than not at all (they are
meant to be public interfaces after all) but I agree the naming is not
optimal. I think it should suffice to make it clear in the docstrings that
they should be used e.g. with `'eeg' in inst` or `inst[...]` as opposed to
the builtin function names themselves.

Eric89GXL, 2016-08-02:
And I also agree first_samp should be better documented. Basically it
should be used/incorporated with events and annotations (for historical
reasons only AFAIK) and nowhere else.

christianbrodbeck, 2016-08-03:
Oh, by private API I did not mean `__getitem__`, I meant `Raw._data`

agramfort, 2016-08-03:
oh boy... looks like @wmvanvliet wrote this back then.

let me do a pass now.

regarding the handling of first_samp the alternative was to have
potentially raw[:, 0] return an empty array. What I implemented, ie
not use first_samp, for __getitem__ seemed like the less suprising.
Now the presence of first_samp in events is there for backward
compatibility reasons. It has always been there for -eve.fif file.

agramfort, 2016-08-03:
see https://github.com/mne-tools/mne-python/pull/3486

Eric89GXL, 2016-08-03:
> Oh, by private API I did not mean __getitem__, I meant Raw._data

Argh, that's what I get for reading on mobile and making assumptions about
the links instead of following them...

Yeah that shouldn't be documented

mmagnuski, 2016-08-12:
this should probably be closed :)

closed 2016-08-12 by agramfort