mne-tools/mne-python #3484: ENH: CSP support for n_classes > 2

kingjr, 2016-08-02:
Currently, CSP doesn't support more than two classes, in principle we can do n classes right @alexandrebarachant ?

alexandrebarachant, 2016-08-02:
Yep.
there is an extension of CSP to multiclass using approximate joint diagonalization.
It's actually implemented in my toolbox. For multiclass, selection and interpretation of filters is not straightforward, so sometime people use a one-versus-one or one-versus-rest approach.

It is actually easy to extend the regular CSP code for this usecase. The only requirement is to have a approximate joint diagonalisation algorithm. Since there is none implemented in numpy or scipy, i translated the 3 most popular ones from the matlab code provided with the original papers : https://github.com/alexandrebarachant/pyRiemann/blob/master/pyriemann/utils/ajd.py

they are tested against the matlab implementation, but probably need rewriting to comply with the MNE coding guidelines.

kingjr, 2016-08-02:
thanks @alexandrebarachant 
@Eric89GXL is this a scipy job, or shall we put in here in mne?

kingjr, 2016-08-02:
@alexandrebarachant I'm trying to give it a go. However, for n_classes > 2, there seems to be a normalization step for which we need to compute to average of the covariances matrices; apparently, this can be done in different ways e.g. metric='riemann': https://github.com/alexandrebarachant/pyRiemann/blob/master/pyriemann/spatialfilters.py#L244

Am I understanding it right? Is this a blocker you think? Shall we support a single metric option for now, and refer to your toolbox otherwise?

alexandrebarachant, 2016-08-02:
@kingjr for CSP, the intra-class covariance matrice is obtained by using an arithmetic (euclidean) average of the covariance matrices of each trial. It might be sub-optimal and you can improve the CSP using other metric for the mean. 

you can check this two papers [here](https://hal.archives-ouvertes.fr/file/index/docid/602686/filename/Barachant_MMSP_2010_final.pdf) and [here](https://hal.inria.fr/hal-01182728/document).  For high dimension / high SNR, the euclidean metric is generally quite good, and sometimes optimal. when the SNR decrease, it help to move toward the harmonic mean.

My opinion is that it does not improve things that much. Riemannian metric is a bit more robust due to the logarithmic behavior, but we could just stick with the classical average.

kingjr, 2016-08-02:
Thanks, so I'll just do `np.mean(covariances, axis=0)`, these considerations are slightly beyond this issue IMO.

alexandrebarachant, 2016-08-02:
yep, switch to the arithmetic average. i expose this option in my toolbox because i can, but i actually almost never used it.

kingjr, 2016-08-02:
"because i can" ^__^

closed 2016-08-06 by agramfort