mne-tools/mne-python #3499: LinAlgError: SVD did not converge

mdclarke, 2016-08-08:
@Eric89GXL 

Im getting LinAlgError: SVD did not converge errors now on files I didn't before when running maxwell_filter. Tried changed the st_duration, but nothing works. Any ideas?

Eric89GXL, 2016-08-08:
Argh, we have to wait for Intel to release a real (proper) fix for this. In the meantime you can try updating to SciPy 0.18. Of course that isn't compatible with a Mayavi installation (it will break it), so you might have to have two environments, one that you use with SciPy 0.18 to work around this error, and another that you use for Mayavi.

mdclarke, 2016-08-08:
OK thanks. Any ideas when Intel will fix this? :)

Eric89GXL, 2016-08-08:
No idea. Feel free to look here for more info, though:

https://software.intel.com/en-us/forums/intel-distribution-for-python/topic/628049

closed 2016-08-08 by Eric89GXL