mne-tools/mne-python #3512: Need two key presses in raw plot

cbrnr, 2016-08-12:
When I plot my raw EEG with `raw.plot()`, the keyboard controls do not work as expected. For example, to scroll one page to the right I press the right arrow key. Nothing happens - I need to press another key until the plot gets updated and shows the next page. I've tested this on Mac OS X with the latest Anaconda distribution in IPython (with and without `%matplotlib`).

mmagnuski, 2016-08-14:
I'm not on a mac but heard from other people that they are also experiencing this problem.

Eric89GXL, 2017-02-14:
@cbrnr is this still a problem with latest `matplotlib`? Have you tried different backends, e.g. MaxOSX vs pyqt4 / pyqt5?

cbrnr, 2017-02-15:
You're right, this seems to be specific to the `macosx` backend. Everything works fine with `Qt5Agg`.

Eric89GXL, 2017-02-15:
Okay, closing as upstream bug. @cbrnr if you have time, can you add an entry to `doc/faq.rst` about OSX `matplotlib` backends?

cbrnr, 2017-02-15:
OK sure! Although I'd appreciate it if someone else reproduced this behavior first - not that it is just me and my specific setup...

Eric89GXL, 2017-03-21:
It sounds like @mmagnuski knows you're not alone in this. Hopefully the MPL people fix it soon.

closed 2017-03-21 by Eric89GXL

cbrnr, 2017-03-21:
Is there a corresponding upstream issue?

Eric89GXL, 2017-03-21:
I'm not entirely sure, maybe one of these (with a possible fix)?

https://github.com/mne-tools/mne-python/issues/2450
https://github.com/matplotlib/matplotlib/pull/6178

Feel free to look further. If there isn't a good one, feel free to make a minimal example for them

Eric89GXL, 2017-03-21:
Ignore that first link, bad copy-paste. But the MPL PR does have some linked issues IIRC

cbrnr, 2017-03-21:
Thanks - I'll see if this issue fits in there. I wouldn't want to forget about it though, because I assume that many MNE users work on a Mac. Maybe MNE should tell them to switch to the Qt5Agg backend when they start up the viewer?