mne-tools/mne-python #3515: issues with animate_topomap in interactive mode

mmagnuski, 2016-08-13:
using `animate_topomap matplotlib` when in interactive mode (`%matplotlib`) does not work very well (at least on windows):
* "Initializing animation..." is printed every time the animation goes back to the beginning - and the whole animation seems to be recomputed from scratch (I need to wait for a few seconds before it restarts)
* clicking the topo does not stop the progression of time - only turns off redrawing. If I click the topo, and wait a few seconds, and then click again - the animation is a number of frames ahead (proportional to the time being paused

agramfort, 2016-08-13:
In a notebook?

mmagnuski, 2016-08-13:
In Spyder, but I can check the notebook.

agramfort, 2016-08-13:
ok it's bad then. @jaeilepp please look when you're back

Eric89GXL, 2017-04-18:
@mmagnuski do you still experience this? I suspect it could just be a transient `matplotlib` bug.

larsoner, 2021-02-19:
Closing since last I checked it worked okay, so probably an old mpl bug

closed 2021-02-19 by larsoner