mne-tools/mne-python #3516: Sphere fitting from digitization points is way too big in `standard_1005` montage

chrismullins, 2016-08-13:
Minimum example:

```python
import mne
import matplotlib.pyplot as plt
import numpy as np
import os.path as op
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'

#   Setup for reading the raw data
raw = mne.io.Raw(raw_fname, preload=True)

# Create the montage, and convert to a DigMontage
tms_montage = mne.channels.read_montage(kind='standard_1005')
dig_montage = mne.channels.DigMontage(hsp=tms_montage.pos, point_names=tms_montage.ch_names, hpi=np.zeros((0,3)), elp=np.zeros((0,3)))

raw.set_montage(dig_montage)
sphere_model = mne.bem.make_sphere_model(r0 = 'auto', head_radius = 'auto', info=raw.info)
```

Produces runtime warnings:
```
Estimated head size (97518.6 mm) exceeded 99th percentile for adult head size
```
and
```
(X, Y) fit (912.7, -17513.4) more than 20 mm from head frame origin
Origin head coordinates:      912.7 -17513.4 406.2 mm
Origin device coordinates:    912.7 -17513.4 406.2 mm
```

However the center calculated from the montage centroid seems pretty close, at least in two dimensions:
```python
x_pos = [p[0] for p in tms_montage.pos]
y_pos = [p[1] for p in tms_montage.pos]
z_pos = [p[2] for p in tms_montage.pos]
montage_centroid = np.array([sum(x_pos) / len(x_pos), sum(y_pos) / len(y_pos), sum(z_pos) / len(z_pos)])
print("Montage centroid: {}".format(montage_centroid))
print("Sphere model center: {}".format(sphere_model['r0']))
```
gives
```
Montage centroid: [  0.67142197 -16.51944075  20.65025723]
Sphere model center: [  0.91271914 -17.51340185   0.40618967]
```

I have only tried this with `standard_1005` points, and creating the montage myself from a csv file.  Looks like a scaling problem, mm being interpreted as meters somewhere maybe (since ~97mm would be a reasonable head size).

agramfort, 2016-08-15:
indeed I suspect a unit pb in tms_montage.pos

can you make sure the pos is in meters?

chrismullins, 2016-08-15:
According to [standard_1005.elc](https://github.com/mne-tools/mne-python/blob/master/mne/channels/data/montages/standard_1005.elc#L3) the units are millimeters.

Eric89GXL, 2016-08-16:
We should probably change whatever under the hood reads that file to respect the "units" entry, that should fix this problem

closed 2016-08-19 by Eric89GXL