mne-tools/mne-python #3519: Create reader for Eximia Nexstim data files (.nxe/.nxa/.nxi)

chrismullins, 2016-08-15:
I imagine the same API would make sense, something like `mne.io.read_raw_nexstim`.

As a followup, here [a tutorial](http://www.fieldtriptoolbox.org/tutorial/tms-eeg) on dealing with TMS-EEG datasets with the fieldtrip toolbox, might be a good format to mimic for a tutorial eventually.

Eric89GXL, 2016-08-15:
I can try to do this for 0.13. Can you share a link to the file that does not work?

Once the reader is implemented, can you work on a tutorial?

chrismullins, 2016-08-15:
> Can you share a link to the file that does not work?

I'll share via email for now -- it's probably fine but these aren't my data.

> Once the reader is implemented, can you work on a tutorial?

Absolutely, I'll make an issue for this and assign it to myself.

Eric89GXL, 2016-08-26:
@agramfort do you have any of these files, or know anyone who uses them?

agramfort, 2016-08-27:
I've never worked with such files

fraimondo, 2017-07-13:
As usual, I need to open a file which is not fully supported by mne python (yet).

I checked out @Eric89GXL  eximia branch and updated it to MNE master.

I tested one of the files that I have. Apparently the data is in uV, and it needs to be scaled to V to MNE.

Here's a screenshot on how the following code works:

```Python
raw = mne.io.read_raw_eximia('/Users/fraimondo/data/tms_eeg/subjects/test1/20170711_085238_12_17_coma_KM_7.nxe', preload=True)
raw.filter(1, 30)
raw._data *= 1e-6
raw.plot()
```

[image]

Trigger 2 and gain are disconnected. While Trigger 1 has trains of 5 stimulations.

Everything looks good from this side, besides the scaling issue. I wanted to fix it, but I don't want to mess with @Eric89GXL tidy code.

agramfort, 2017-07-13:
yes please take over and make sure we have a tiny file in testing we can
use on travis.

thx Fede !

fraimondo, 2017-07-13:
Ok, that will be for after holidays.

for now, I just made a clean start on mne master: https://github.com/fraimondo/mne-python/commits/add_eximia

Eric89GXL, 2017-07-13:
Why start from scratch given my branch (almost?) works?

Eric89GXL, 2017-07-13:
If you can instead provide some small test file, plus ideally the data
values from some other software, I can update my branch and it should be
ready for testing and merge

fraimondo, 2017-07-13:
Indeed what I did is to take your branch and update it to mne master so I can test with working code.

I don’t have access to the eximia system, so I only have the nxe files. I guess tomorrow I will be able to go and try to export the files in a different format. Also to crop this huge file I have and share a small test file.

larsoner, 2018-04-16:
Done in #5043

closed 2018-04-16 by larsoner