mne-tools/mne-python #3525: "Morph" dipole to MNI space

agramfort, 2016-08-17:
we have a vertex_to_mni  function for source space based inverse methods

it would be great to have the same thing for dipole fits. E.g.

dipole.to_mni() -> returns new Dipole instance where dipole.pos is in MNI coordinates.

thoughts?

Eric89GXL, 2016-08-19:
Probably requires something like #3527

Eric89GXL, 2017-02-16:
@agramfort do you have an API in mind? I'm thinking:
```
dip.to_mni(subject)
```
Where `subject` is a FreeSurfer subject name.

Unfortunately the dipole fits are not necessarily constrained to a cortical surface, so I think we need to use volumetric / MRI coregistration instead of surface-based coregistration to get there. Agreed?

Eric89GXL, 2017-02-16:
... oh wait, I think each FreeSurfer subject already has an MNI transform. So this should be very simple actually, assuming I have the API right.

agramfort, 2017-02-17:
yes it MNE would be great to we could overlay this with the FSL MNI brain
template

dip.to_mni works for me

vagechirkov, 2021-03-15:
I would like to try to solve this issue (sprint 2021) :)

sappelhoff, 2021-03-19:
closed in #9091

closed 2021-03-19 by sappelhoff