mne-tools/mne-python #3527: ENH: Anatomical template warping

Eric89GXL, 2016-08-19:
I'm starting to read a bit about anatomical template warping, and it looks like there are some standard methods like [TPS warping]  (http://onlinelibrary.wiley.com/wol1/doi/10.1002/hbm.20171/full) or warping based on [spherical harmonics] (http://ieeexplore.ieee.org/xpl/articleDetails.jsp?arnumber=966602) that can be used. Any thoughts on the best ones to use?

@ftadel you use the TPS warping technique in BrainStorm, right?

ftadel, 2016-08-19:
Hi Eric,

We use code written by John Mosher and John Ermer: we start by computing a smooth surface based on the digitized head points (using spherical harmonics), then we compute a deformation field based on the two surfaces (the one from the MRI and the one from the Polhemus) and apply it on the surfaces and the MRI.
It is done in function brainstorm3/toolbox/anatomy/bst_warp.m
(yes, I'll move my code base to github soon :)

The corresponding tutorial: http://neuroimage.usc.edu/brainstorm/Tutorials/TutWarping

Francois

Eric89GXL, 2016-08-22:
Thanks @ftadel. @christianbrodbeck would this be useful for times when you want to use a surrogate? Any thoughts?

christianbrodbeck, 2016-08-22:
Hm yes warping might potentially improve the localization for surrogates if the head shape is accurate. Although the advantage over scaling would depend on the head shape reflecting brain shape. Do we have an idea how well brain shape is correlated with head shape?

Eric89GXL, 2016-08-22:
IIRC both of the papers linked above do tests related to that and show increases in inverse accuracy

PR #3613 (2016-10-18): MRG: Add subject warping
This adds support for warping between subjects using spherical harmonics. The 0th order case is fitting spheres for two sets of points and using translation plus radius-scaling to transform; order 1 uses [something like an oblate spheroid, etc.] (https://en.wikipedia.org/wiki/Spherical_harmonics)

~~This can create stuff like this (the stuff of Matti-related nightmares), for order 4 trying to fit a true surface using just dig points (orig surface is gray, dig in dark gray, spherical transformation/deformation in purple, resulting surface in green; [code in a gist] (upd [... cut]
closed 2016-10-18 by agramfort