mne-tools/mne-python #3531: Passing times of interest to time-frequency functions

mmagnuski, 2016-08-25:
What do you think about adding the ability to specify timepoints of interest in high-level time-frequency functions (like `tfr_morlet`)?
Currently one would have to use for example `epochs.time_as_index(times_of_interest)` and then create a slice to pass to `decim`. It's works ok for me but is not very intuitive for newcomers.
If you are :+1: then the API could be:
* passing a slice constructed from floats to `decim` would be interpreted as time in seconds and translated to relevant integer slice
* a separate parameter, for example `times` or `toi` (after fieldtrip from times of interest) that would take an array-like of floats (or float slice)

I prefer option 2 because it is more intuitive to use `times` than `decim` to select relevant timepoints.

kingjr, 2016-08-25:
`decim` could accept a slice of floats, in which case it looks for the data `tmin` `sfreq` etc.

agramfort, 2016-08-27:
you want to compute/visualize TFRs with irregular time instants?

mmagnuski, 2016-08-27:
No, just allow to use time values instead of samples.

agramfort, 2016-08-29:
then I am not sure about the use case. You want time instants that are not
subsamples of times array? like do some interpolation?

mmagnuski, 2016-08-29:
No, just translate float slices to int slices (seconds -> samples). This may lead to accumulating offsets due to rounding, so might be better to translate float slice to an array of indices (similar to `time_as_index`) etc.
This could be separate from `decim` however, similar to the way `foi` works in fieldtrip: `time=np.arange(-0.25, 0.501, 0.05)`. I think it would be more convenient for newcomers.

agramfort, 2016-08-29:
I am not convinced :) but I can be wrong...

to me it would open the API too much

your TFRs does not depend much on fine grained time selection
due to limited time resolution. You can do a pretty good job
with crop and proper decim.

any thoughts from others?

mmagnuski, 2016-08-29:
> your TFRs does not depend much on fine grained time selection
due to limited time resolution. You can do a pretty good job
with crop and proper decim.

My line of thought was a bit different: let's say I have a pretty long signal (long epochs or raw) and I want to decimate the tfr so that I have time like `0.:15.:0.05`. If I translate that to int slice (timesamples) in some cases the rounding will lead to for example the step that should be at 10s (along the logic of the float slice) be at 9.5s. 
Anyway - I am fine with crafting decim slice myself and being aware of what I'm doing, but it's common to use time values (like `toi` in fieldtrip) and it is more intuitive than `decim` in samples.

agramfort, 2016-08-29:
I am willing to renegotiate next time pb shows up :)

mmagnuski, 2016-08-29:
Does it mean no to 1) `time` param, 2) float slice in `decim` or 3) translating float `decim` to array of ints (instead of slice of inst)? :)

agramfort, 2016-08-29:
I would say no indeed

mmagnuski, 2016-08-29:
ok :( / :)

kingjr, 2016-08-30:
you can close the issue ;)

mmagnuski, 2016-08-30:
Farewell my dear issue then... :snowman:

closed 2016-08-30 by mmagnuski