mne-tools/mne-python #3536: ENH: make _get_ica_map a public method of ICA

jona-sassenhagen, 2016-08-27:
We ever so often get people wondering how to retrieve ICA maps as numpy arrays, and I keep referring them to mne.preprocessing.ica._get_ica_map. It would probably be easier to make it a public method of ICA. Opinions?

agramfort, 2016-08-27:
+1 but maybe get it as an EvokedArray to keep channel names?

ica.get_components ?

jona-sassenhagen, 2016-08-27:
> get it as an EvokedArray to keep channel names

For the use cases I've come across, that'd be unnecessary and most of the time users would have to do ica.get_components().data in that case. See e.g. #3460, #2404 and the mailing list one.

@mmagnuski opinion?

mmagnuski, 2016-08-27:
Yes, in this case I also think simpler is better - the channel names are in ica object and in most cases one just needs to get n_channels by n_components arrays so exactly the way `_get_ica_map` works now.

jona-sassenhagen, 2016-08-28:
@agramfort okay? `ica.get_components()` to simply return an array?

We can also give it a single parameter to optionally return either an Evoked object, or e.g. a pandas dataframe.

mmagnuski, 2016-08-28:
or an xarray ;)
probably would be nice to have a param to choose between getting mixing or unmixing matrix?

agramfort, 2016-08-29:
yes fine with array if you can make a valid EvokedArray with something like

mne.EvokedArray(ica.get_components, ica.info, tmin=0.)

to be added as a test.

PR #3818 (2016-12-13): WIP add get_components method
Remake of #3573 cause we had a lot of discussion.

closes #3536

@agramfort @mmagnuski @agostontorok
closed 2016-12-13 by agramfort