mne-tools/mne-python #3538: Document EEG reference

wmvanvliet, 2016-08-29:
We keep getting questions of users that are confused with MNE's EEG referencing behavior. This is not surprising, as the documentation does not cover this topic. At the very least, we need to add a note that for EEG data, a reference **must** be specified at all times. If the user does nothing, MNE will add an average reference. If another reference is desired, the user should use the `mne.io.set_eeg_reference` function to specify this.

jasmainak, 2016-08-29:
hmm ... but did you look at the line pointed out in the mailing list? It seems to be a bug rather than a doc issue

jasmainak, 2016-08-29:
Basically, even if you remove the ref proj, it still gets added when creating the Epochs object because the avg_ref param is not passed to it.

agramfort, 2016-08-29:
+1 to add such documentation. Any suggestion where?

also the more I get questions about EEG reference the more I feel that:

mne.io.set_eeg_reference

is not easy to find and the io module is clearly not a naturally place to look.

2 options:

- move mne.io.set_eeg_reference to mne.set_eeg_reference

- add a set_eeg_reference method to the data containers. I know I was opposed to this to keep the number of methods low but it's true that it's a quite general use case...

jona-sassenhagen, 2016-08-29:
A method seems natural to me.

wmvanvliet, 2016-08-29:
> Basically, even if you remove the ref proj, it still gets added when creating the Epochs object because the avg_ref param is not passed to it.

Yes, and in other places too. The solution we implemented a while ago is to use the `mne.io.set_eeg_reference` function. It sets the `custom_ref_applied` flag in the `info`, which prevents MNE from automatically adding any further referencing.

Just removing the ref proj. will cause MNE to assume no referencing has been applied and add its own. For some steps, you can explicitly tell MNE not to do it (by setting `avg_ref=False`), but not for all. Perhaps we should modify the code to allow the user to set this for all cases that may add an average reference. However, working like this is extremely annoying for the user, who must remember at every step to add the `avg_ref=False` flag.

jasmainak, 2016-08-29:
hmm ... I did not know there was another way to prevent average referencing. It's not super clear to me how it's supposed to work and how these different flags interact with each other. A while ago I discovered that doing `proj=False` does not prevent average reference from being applied even though average reference is treated as a proj in MNE. You have to explicitly set `avg_ref=False`. Now, you're telling me that there's another way to disable the average reference. To add to the confusion, you have `active` flag which tells you if a proj has been applied and now there's also a `custom_ref_applied` flag which tells you if EEG ref has been applied?

wmvanvliet, 2016-08-29:
> and now there's also a custom_ref_applied flag which tells you if EEG ref has been applied?

Yup. Hence, the dire need to document this stuff.

Eric89GXL, 2016-08-29:
This is really a dup of #3410, but I'll close that one since this one has a richer discussion.

Both of these are almost a #2727, which highlights how we have done some inconsistent things with `add_egg_ref` for a long time.

I agree with @agramfort (now) that we should add a method. But I would go a step further and make it the only way to add an EEG reference. It's more pythonic than having `add_eeg_ref` args, sometimes defaulting to `True`, sporadically placed in our code -- which is what we have now. So I'd actually really prefer deprecating all `add_eeg_ref` args. Perhaps more importantly, it forces people to be more explicit about when the average reference gets added. We already have safeguards against inversion without average ref, so we should be safe to do this.

Eric89GXL, 2016-08-29:
> For some steps, you can explicitly tell MNE not to do it (by setting avg_ref=False), but not for all. Perhaps we should modify the code to allow the user to set this for all cases that may add an average reference. 

In other words, I really think we should deal with this by getting rid of the `avg_ref` parameters everywhere, and emitting errors if they try to do something that requires avg ref but it isn't present. It's a much cleaner / one-way / explicit approach than having `avg_ref`'s all over the place (some of which defaulting to `True` that people might not notice).

wmvanvliet, 2016-08-29:
The `set_eeg_reference` function comes as part of a set of tools related to EEG references, which also includes `set_bipolar_reference` and `add_reference_channels`. I actually don't mind them being in the `mne.io` module, since I think of them as being part of loading the Raw data. Although +1 for exposing them to the top level `mne` module.

agramfort, 2016-08-29:
any objection with @Eric89GXL?

what would be a clear step by step plan?

bonus question : who has time to take this over?

Eric89GXL, 2016-08-29:
I can implement it. Plan would be something like:

1. 0.13:
    1. deprecate `add_eeg_ref=True` -> `add_eeg_ref=False`
    2. Change all examples to use `add_eeg_ref=False` plus preferred referencing methods
    3. Update docs to say when refs will be applied in 0.13 and 0.14
2. 0.14
    1. deprecate `add_eeg_ref`
    2. Change all examples not to use `add_eeg_ref` at all

@agramfort we should probably make all of `set_eeg_reference`, `set_bipolar_reference`, and `add_reference_channels` either methods, or not. So okay to add these three?

PR #3545 (2016-09-15): [MRG+1]: Deprecate add_eeg_ref parameters
This PR basically acts as an API proposal for #3538 and #2727. @agramfort @dengemann and others feel free to have a look.

The one API thing that's missing is making `set_eeg_reference` a method of `Raw`, `Epochs`, and `Evoked`, which I think we should do. I'm okay with not doing the bipolar or `add_reference_channels` since they are used less often, and we can link to them in `See Also`. Is that okay with you @wmvanvliet?

I still need to update all examples along the lines of the one I did so far, as well as all tests that load raw data (ouch).

Clos [... cut]
closed 2016-09-15 by agramfort