mne-tools/mne-python #3539: MNE Python for fNIRS?

mcaffini, 2016-08-29:
Hi all,

I successfully tested some of the **MNE Python** methods and classes in my **fNIRS** data processing workflow.
I wonder if anyone else is interested in this and whether it's worth to write some code in this direction. I can also swallow the bitter pill and keep labeling my fNIRS data as "eeg", but maybe we could seize the day to start something useful. Happy to share more of what I did if needed.

Let's discuss it! :-)

Eric89GXL, 2016-08-29:
I see no problem with asking to add a fNIRS channel type to the FIF tags and making the appropriate adjustments in our code. @agramfort @jaeilepp do you agree?

Eric89GXL, 2016-08-29:
@mcaffini do you have some small data file that you can share? What format is it?

jaeilepp, 2016-08-29:
Sure, why not.

Eric89GXL, 2016-08-29:
Can you ask the Neuromag folks?

agramfort, 2016-08-29:
fine with me.

mcaffini, 2016-09-01:
Dear all, sorry for the late reply.
@Eric89GXL I use the ISS Imagent. fNIRS raw data come from different technologies (continuous wave, frequency domain, time domain) and each instruments has its own format, but they all eventually get transformed into oxyhemoglobin (HbO) and deoxyhemoglobin (HbR) concentrations (this transformation process is different for different technologies). Maybe we skip this step and we directly start from a numpy array containing the concentrations? This way we would only need two tags like fNIRS-HbO and fNIRS-HbR...

jaeilepp, 2016-09-01:
> Maybe we skip this step and we directly start from a numpy array containing the concentrations?

Sounds reasonable. We probably don't want to provide the 'preprocessing' steps for this transformation anyway.

jaeilepp, 2016-09-01:
Actually, it sounds like we would need a tag for fNIRS channel type and for two units (HbO and HbR). Am I right?

mcaffini, 2016-09-01:
What do you mean with unit? If you have in mind each channel having two independent tracks/signals (HbO and HbR), that sounds good!

Eric89GXL, 2016-09-01:
Physical units, preferably of the SI variety -- for EEG it's V (commonly viewes as uV) and for MEG it's T and T/m (more commonly viewed as fT / fT/cm).

mcaffini, 2016-09-01:
OK, then for fNIRS is molar concentration (M=mol/l=mol/dm3, usually as μM, proper SI unit would be mol/m3). Each channel gives two independent signals (HbO and HbR) with the same unit (μM). How do we refer to each one separately?

jaeilepp, 2016-09-01:
Ok, yeah. I'm thinking that maybe we need just one channel type with two kinds (like we have with MEG channels mag and grad). Makes sense? Although it might be the easier to have just two types. Not sure which is better.

mcaffini, 2016-09-01:
One type with two kinds makes more sense to me. That way you can pick all HbO with 'hbo', all HbR with 'hbr', or both with 'fnirs' (no caps to be consistent with the other tags?), the same way you pick magnetometers with 'mag', gradiometers with 'grad', ... or everything MEG with 'meg'. Adding total hemoglobin (HbT or hbt) is trivial (but not necessary) being the algebraic sum of HbO and HbR.

In http://martinos.org/mne/stable/generated/mne.pick_types.html would it look like this?

[...]
**meg** : bool or string
If True include all MEG channels. If False include None. If string it can be 'mag', 'grad', 'planar1' or 'planar2' to select only magnetometers, all gradiometers, or a specific type of gradiometer.

**fnirs**: bool or string
If True include all fNIRS channels. If False include None. If string it can be 'hbo', 'hbr', 'hbt' to select only oxyhemoglobin, deoxyhemoglobin, or total hemoglobin (algebraic sum of hbo and hbr).
[...]

agramfort, 2016-09-01:
yes that sounds good to me

agramfort, 2016-09-15:
@jaeilepp what did we converge on regarding fif constants?

jaeilepp, 2016-09-15:
No response from Elekta yet.

jaeilepp, 2016-09-16:
Here they are:
```
fnirs_hbo 300  "FNIRS oxyhemoglobin"
fnirs_hbr 301  "FNIRS deoxyhemoglobin"
fnirs 1100 "Functional near-infrared spectroscopy"
FIFF.FIFFV_FNIRS_CH     = 1100
```

agramfort, 2016-09-16:
cool

thx @jaeilepp

let's get this merged

closed 2016-09-22 by Eric89GXL

agramfort, 2017-04-15:
@mcaffini there is a new public dataset with EEG + NIRS:

https://www.depositonce.tu-berlin.de/handle/11303/6271

maybe a good way to have a first example of NIRS with MNE?

mcaffini, 2017-04-15:
Thanks @agramfort very interesting, I'll give it a shot.
I have completed my fNIRS analysis with MNE and I have also coded the source space reconstruction using a Monte Carlo method for the forward solution.
Still interested in a tutorial for the MNE website?

dengemann, 2017-04-15:
Yes, this sounds really cool. Go for it!

agramfort, 2017-04-16:
>
> Thanks @agramfort <https://github.com/agramfort> very interesting, I'll
> give it a shot.
> I have completed my fNIRS analysis with MNE and I have also coded the
> source space reconstruction using a Monte Carlo method for the forward
> solution.
> Still interested in a tutorial for the MNE website?
>
I don't know about the forward code I would to see it but basic raw data viz
would already be neat I think.

mcaffini, 2017-05-09:
I prepared a Jupyter notebook tutorial with basic fNIRS data analysis, [here](https://github.com/mcaffini/mne-python-fnirs). Of course comments and suggestions are more then welcome. :-)

agramfort, 2017-05-10:
we cannot really comment code with a rendered notebook

could be done with public data available from a website so we can
write a data fetcher? maybe not start from a custom .mat file which
generates a lot of boiler plate code which clutters the example/tutorial

mcaffini, 2017-05-10:
@agramfort the notebook as well as the npy file with data should be downloadable for testing from my repo. The provided dataset comes from my lab and is real data, but we can change dataset and fetch it online, no problem.
Btw, I don't see any boilerplate code (apart from numpy.load :-D).

agramfort, 2017-05-10:
can you make a tutorial script in a PR so we can comment line by line?