mne-tools/mne-python #3541: Finding bad ecg components with ECG channel

jasmainak, 2016-08-29:
I am trying to find bad ecg components in ICA. I have an EOG channel and I want to use it to find the components. Unfortunately, I'm facing issues of missing MEG channels even though my data has only EEG. Here's a small script to reproduce the problem:

```py
import mne
from mne.preprocessing import ICA, create_ecg_epochs
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'

raw = mne.io.read_raw_fif(raw_fname, preload=True)
raw.info['projs'] = []
raw.set_channel_types({'MEG 2641': 'ecg'})
raw.pick_types(meg=False, eeg=True, exclude='bads', stim=True, ecg=True)
raw.filter(1, 30, method='iir')

events = mne.find_events(raw, stim_channel='STI 014')
epochs = mne.Epochs(raw, events, event_id=None, tmin=-0.2, tmax=0.5)

ica = ICA(method='fastica').fit(epochs)
ecg_epochs = create_ecg_epochs(raw, ch_name='MEG 2641', tmin=-.5, tmax=.5)
ecg_inds, scores = ica.find_bads_ecg(ecg_epochs)
```

jaeilepp, 2016-08-30:
``create_ecg_epochs`` has an option ``keep_ecg``. Use that.

jasmainak, 2016-08-30:
oh, yes. Thanks!

jasmainak, 2016-08-30:
This works (thanks @jaeilepp!) but reopening issue as there are still some problems, namely:

1. the documentation of keep_ecg needs fixing. It gives the impression that ECG channel is dropped only if it's created synthetically. But this is clearly not the case. It is also dropped when a channel name is provided or when an ECG channel exists
2. the documentation for ch_name also needs to be fixed. It says that a synthetic channel is created from cross channel average if it is None. However, in my case, where I have an ECG channel, it is in fact using that channel and not creating a synthetic channel.
3. finally, the default in ica.find_bads_ecg for method='ctps' is 0.25. This appears to be very low. Maybe we should update this default.

agramfort, 2016-08-30:
PR welcome for docstrings

for 3/ it would need quantified arguments

closed 2016-09-02 by jasmainak